YosysHQ/yosys #959: read_verilog prints "Executing Verilog-2005 frontend." for every the source file in the design.

ravenexp, 2019-04-26:
It would be nice if the message also contained the source file name.
Looking at the source, the filename argument seems to be available right at the point of printing of the header:

https://github.com/YosysHQ/yosys/blob/408161ea3af78c747b9d45cd6482f2e4d9170085/frontends/verilog/verilog_frontend.cc#L245

cliffordwolf, 2019-04-26:
The filename is only readily available after the call to `extra_args()`. And right after that we write the file name to the log file:

https://github.com/YosysHQ/yosys/blob/408161ea3af78c747b9d45cd6482f2e4d9170085/frontends/verilog/verilog_frontend.cc#L416-L419

ravenexp, 2019-04-26:
Oh, so the "filename" argument of execute() does not actually contain a filename?
But it is then somehow reused as a local variable and gets initialized via extra_args()?
I'm confused.

It's somewhat annoying that yosys prints the same header line "Executing Verilog-2005 frontend." around 150 times while loading my design.
It's uninformative and it can't be hidden with -v option since it's a level 1 header.

cliffordwolf, 2019-04-30:
> Oh, so the "filename" argument of execute() does not actually contain a filename?

There are different ways how a front-end can get called. Either with a filename in `filename` and the already open file in `&f`, or with a filename in the arguments. And in the latter case `extra_args` will open the file for us and update `filename` and `f` accordingly.

> It's uninformative and it can't be hidden with -v option since it's a level 1 header.

I've now move the log_header to a point where the filename is known in 84f3a79, and included the filename in the log header. It's a bit idiosyncratic compared to what we do in other passes, but I agree that the log output is significantly more useful this way.

closed 2019-04-30 by cliffordwolf

ravenexp, 2019-05-02:
Thank you for your prompt response.
This change solves my issue perfectly.

commit 84f3a796 (2019-04-30): Include filename in "Executing Verilog-2005 frontend" message, fixes #959