YosysHQ/yosys #961: Segfault for big cascaded-if

eddiehung, 2019-04-26:
## Steps to reproduce the issue

`yosys -p proc test.v`

```
module mux_if_33_8 #(parameter N=33, parameter W=8) (input [N*W-1:0] i, input [$clog2(N)-1:0] s, output reg [W-1:0] o);
always @*
    if (s == 0) o <= i[0+:W];
    else if (s == 1) o <= i[1*W+:W];
    else if (s == 2) o <= i[2*W+:W];
    else if (s == 3) o <= i[3*W+:W];
    else if (s == 4) o <= i[4*W+:W];
    else if (s == 5) o <= i[5*W+:W];
    else if (s == 6) o <= i[6*W+:W];
    else if (s == 7) o <= i[7*W+:W];
    else if (s == 8) o <= i[8*W+:W];
    else if (s == 9) o <= i[9*W+:W];
    else if (s == 10) o <= i[10*W+:W];
    else if (s == 11) o <= i[11*W+:W];
    else if (s == 12) o <= i[12*W+:W];
    else if (s == 13) o <= i[13*W+:W];
    else if (s == 14) o <= i[14*W+:W];
    else if (s == 15) o <= i[15*W+:W];
    else if (s == 16) o <= i[16*W+:W];
    else if (s == 17) o <= i[17*W+:W];
    else if (s == 18) o <= i[18*W+:W];
    else if (s == 19) o <= i[19*W+:W];
    else if (s == 20) o <= i[20*W+:W];
    else if (s == 21) o <= i[21*W+:W];
    else if (s == 22) o <= i[22*W+:W];
    else if (s == 23) o <= i[23*W+:W];
    else if (s == 24) o <= i[24*W+:W];
    else if (s == 25) o <= i[25*W+:W];
    else if (s == 26) o <= i[26*W+:W];
    else if (s == 27) o <= i[27*W+:W];
    else if (s == 28) o <= i[28*W+:W];
    else if (s == 29) o <= i[29*W+:W];
    else if (s == 30) o <= i[30*W+:W];
    else if (s == 31) o <= i[31*W+:W];
    else if (s == 32) o <= i[32*W+:W];
    else o <= {W{1'bx}};
endmodule
```

## Expected behavior

No SIGSEGV.

## Actual behavior

```
(gdb) bt
#0  0x0000000000620d58 in Yosys::AST_INTERNAL::ProcessGenerator::processAst (this=0x7fffffffc1f0, ast=<optimized out>) at frontends/ast/genrtlil.cc:551
#1  0x0000000000620570 in Yosys::AST_INTERNAL::ProcessGenerator::processAst (this=0x7fffffffc1f0, ast=0xfdd140) at frontends/ast/genrtlil.cc:438
#2  0x0000000000620946 in Yosys::AST_INTERNAL::ProcessGenerator::processAst (this=0x7fffffffc1f0, ast=0xfd1150) at frontends/ast/genrtlil.cc:513
#3  0x0000000000620570 in Yosys::AST_INTERNAL::ProcessGenerator::processAst (this=0x7fffffffc1f0, ast=0xfd1020) at frontends/ast/genrtlil.cc:438
#4  0x0000000000620946 in Yosys::AST_INTERNAL::ProcessGenerator::processAst (this=0x7fffffffc1f0, ast=0xfc70a0) at frontends/ast/genrtlil.cc:513
#5  0x0000000000620570 in Yosys::AST_INTERNAL::ProcessGenerator::processAst (this=0x7fffffffc1f0, ast=0xfc5330) at frontends/ast/genrtlil.cc:438
#6  0x0000000000620946 in Yosys::AST_INTERNAL::ProcessGenerator::processAst (this=0x7fffffffc1f0, ast=0xfb8490) at frontends/ast/genrtlil.cc:513
#7  0x0000000000620570 in Yosys::AST_INTERNAL::ProcessGenerator::processAst (this=0x7fffffffc1f0, ast=0xfb8360) at frontends/ast/genrtlil.cc:438
#8  0x0000000000620946 in Yosys::AST_INTERNAL::ProcessGenerator::processAst (this=0x7fffffffc1f0, ast=0xfac890) at frontends/ast/genrtlil.cc:513
#9  0x0000000000620570 in Yosys::AST_INTERNAL::ProcessGenerator::processAst (this=0x7fffffffc1f0, ast=0xfac760) at frontends/ast/genrtlil.cc:438
#10 0x0000000000620946 in Yosys::AST_INTERNAL::ProcessGenerator::processAst (this=0x7fffffffc1f0, ast=0xf8fa70) at frontends/ast/genrtlil.cc:513
#11 0x0000000000620570 in Yosys::AST_INTERNAL::ProcessGenerator::processAst (this=0x7fffffffc1f0, ast=0xfdcd00) at frontends/ast/genrtlil.cc:438
#12 0x0000000000620946 in Yosys::AST_INTERNAL::ProcessGenerator::processAst (this=0x7fffffffc1f0, ast=0xfcea10) at frontends/ast/genrtlil.cc:513
#13 0x0000000000620570 in Yosys::AST_INTERNAL::ProcessGenerator::processAst (this=0x7fffffffc1f0, ast=0xfccb70) at frontends/ast/genrtlil.cc:438
#14 0x0000000000620946 in Yosys::AST_INTERNAL::ProcessGenerator::processAst (this=0x7fffffffc1f0, ast=0xfbdd20) at frontends/ast/genrtlil.cc:513
#15 0x0000000000620570 in Yosys::AST_INTERNAL::ProcessGenerator::processAst (this=0x7fffffffc1f0, ast=0xfbbe80) at frontends/ast/genrtlil.cc:438
#16 0x0000000000620946 in Yosys::AST_INTERNAL::ProcessGenerator::processAst (this=0x7fffffffc1f0, ast=0xfb0280) at frontends/ast/genrtlil.cc:513
#17 0x0000000000620570 in Yosys::AST_INTERNAL::ProcessGenerator::processAst (this=0x7fffffffc1f0, ast=0xfb0170) at frontends/ast/genrtlil.cc:438
#18 0x0000000000620946 in Yosys::AST_INTERNAL::ProcessGenerator::processAst (this=0x7fffffffc1f0, ast=0xfa6210) at frontends/ast/genrtlil.cc:513
#19 0x0000000000620570 in Yosys::AST_INTERNAL::ProcessGenerator::processAst (this=0x7fffffffc1f0, ast=0xf936a0) at frontends/ast/genrtlil.cc:438
#20 0x0000000000620946 in Yosys::AST_INTERNAL::ProcessGenerator::processAst (this=0x7fffffffc1f0, ast=0xfde180) at frontends/ast/genrtlil.cc:513
#21 0x0000000000620570 in Yosys::AST_INTERNAL::ProcessGenerator::processAst (this=0x7fffffffc1f0, ast=0xfdcab0) at frontends/ast/genrtlil.cc:438
#22 0x0000000000620946 in Yosys::AST_INTERNAL::ProcessGenerator::processAst (this=0x7fffffffc1f0, ast=0xfd3320) at frontends/ast/genrtlil.cc:513
#23 0x0000000000620570 in Yosys::AST_INTERNAL::ProcessGenerator::processAst (this=0x7fffffffc1f0, ast=0xfcd650) at frontends/ast/genrtlil.cc:438
#24 0x0000000000620946 in Yosys::AST_INTERNAL::ProcessGenerator::processAst (this=0x7fffffffc1f0, ast=0xfc1b40) at frontends/ast/genrtlil.cc:513
#25 0x0000000000620570 in Yosys::AST_INTERNAL::ProcessGenerator::processAst (this=0x7fffffffc1f0, ast=0xfc37e0) at frontends/ast/genrtlil.cc:438
#26 0x0000000000620946 in Yosys::AST_INTERNAL::ProcessGenerator::processAst (this=0x7fffffffc1f0, ast=0xfb87c0) at frontends/ast/genrtlil.cc:513
#27 0x0000000000620570 in Yosys::AST_INTERNAL::ProcessGenerator::processAst (this=0x7fffffffc1f0, ast=0xfb7cd0) at frontends/ast/genrtlil.cc:438
#28 0x0000000000620946 in Yosys::AST_INTERNAL::ProcessGenerator::processAst (this=0x7fffffffc1f0, ast=0xfaea60) at frontends/ast/genrtlil.cc:513
#29 0x0000000000620570 in Yosys::AST_INTERNAL::ProcessGenerator::processAst (this=0x7fffffffc1f0, ast=0xfadf70) at frontends/ast/genrtlil.cc:438
#30 0x0000000000620946 in Yosys::AST_INTERNAL::ProcessGenerator::processAst (this=0x7fffffffc1f0, ast=0xf920c0) at frontends/ast/genrtlil.cc:513
#31 0x0000000000620570 in Yosys::AST_INTERNAL::ProcessGenerator::processAst (this=0x7fffffffc1f0, ast=0xf93e30) at frontends/ast/genrtlil.cc:438
#32 0x0000000000620946 in Yosys::AST_INTERNAL::ProcessGenerator::processAst (this=0x7fffffffc1f0, ast=0xf89ed0) at frontends/ast/genrtlil.cc:513
#33 0x0000000000620570 in Yosys::AST_INTERNAL::ProcessGenerator::processAst (this=0x7fffffffc1f0, ast=0xfde070) at frontends/ast/genrtlil.cc:438
#34 0x000000000061e830 in Yosys::AST_INTERNAL::ProcessGenerator::ProcessGenerator (this=<optimized out>, always=<optimized out>, initSyncSignalsArg=...)
    at frontends/ast/genrtlil.cc:279
#35 0x0000000000615929 in Yosys::AST::AstNode::genRTLIL (this=0xf8ac10, width_hint=<optimized out>, sign_hint=<optimized out>) at frontends/ast/genrtlil.cc:1533
#36 0x00000000005f5336 in Yosys::process_module (ast=0xf88d10, defer=<optimized out>, original_ast=<optimized out>) at frontends/ast/ast.cc:1065
#37 0x00000000005f43d3 in Yosys::AST::process (design=<optimized out>, ast=<optimized out>, dump_ast1=<optimized out>, dump_ast2=<optimized out>,
    no_dump_ptr=<optimized out>, dump_vlog1=<optimized out>, dump_vlog2=<optimized out>, dump_rtlil=<optimized out>, nolatches=<optimized out>,
    nomeminit=<optimized out>, nomem2reg=<optimized out>, mem2reg=<optimized out>, noblackbox=<optimized out>, lib=<optimized out>, nowb=<optimized out>,
---Type <return> to continue, or q <return> to quit---
```

Could this be a stack space issue?

eddiehung, 2019-04-26:
Finding it really hard to reproduce reliably, outside of running my new yosys-bench `equiv_opt` script: https://github.com/YosysHQ/yosys-bench/compare/eddie/equiv

cliffordwolf, 2019-04-30:
> Finding it really hard to reproduce reliably, outside of running my new yosys-bench equiv_opt script

What commands do I need to run exactly in order to reproduce this?

eddiehung, 2019-05-03:
Can't seem to repro at all now, even on my branch. Let's close this for now and I'll re-open it if I run into it again. Sorry for the noise.

closed 2019-05-03 by eddiehung