YosysHQ/yosys #965: read_liberty doesn't support the name_list syntax

porglezomp, 2019-04-27:
I have a liberty file that uses the `pin(name_list)` syntax to define multiple input pins, but yosys doesn't appear to handle this.

## Steps to reproduce the issue

Attempt to `read_liberty` a `.lib` that contains a `pin(name_list)` group.

For example,
```liberty
library(example) {
  cell(NAND) {
    area: 9;
    pin(A, B) { direction: input; }
    pin(Y) { direction: output; function: "!(A & B)"; }
  }
}
```

## Expected behavior

I believe should be read as equivalent to the following:
```liberty
liberty(example) {
  cell(NAND) {
    area: 9;
    pin(A) { direction: input; }
    pin(B) { direction: input; }
    pin(Y) { direction: output; function: "!(A & B)"; }
  }
}
```

## Actual behavior

```
2. Executing Liberty frontend.
ERROR: Can't resolve wire name A.
```

cliffordwolf, 2019-05-01:
cc @trcwm

trcwm, 2019-05-01:
I’ll look into this.

trcwm, 2019-05-02:
FWIW: ABC does not seem to handle 'name_list' arguments either. 

It gives ```ABC: Mio_ParseFormula(): The parser cannot find var "A & B))" in the input var list.``` for the NAND gate.