YosysHQ/yosys #968: Loop variables don't persist following the loop but within the same block

ZipCPU, 2019-04-29:
## Steps to reproduce the issue

Using the following code:

    `default_nettype none
    module(i_clk, jmp, in, out);
      parameter UNUSED=5;
      input wire i_clk, in, jmp;
      output reg [8-1:0] out;

      initial out = 0;
      always @(posedge i_clk)
        if (jmp)
          out <= { out[UNDEFINED-1:0], in };
        else
          out <= { 1'b0, out[UNDEFINED-1:1] };
       // else
       //    out  =0;
    endmodule

If I run this using the yosys commands,

    read -formal test.v
    prep

Yosys crashes on a failed assert telling me nothing about what's going on.

    yosys> read -formal test.v
    1. Executing Verilog-2005 frontend.
    Parsing
## Expected behavior

1. Minimum, I expect a line number showing me where my problem is, so that I might realize I haven't defined UNDEFINED.
2. Because `default_nettype is none, I expect to receive an error before yosys even tries to evaluate this (broken) expression

## Actual behavior

*Please describe how the behavior you see differs from the expected behavior.*

**Important Note:** Nobody will be able to help you and/or fix the issue if you
do not provide sufficient information for reproducing the problem.

cliffordwolf, 2019-04-30:
> I'd expect the resulting Verilog code to look something like, ``assign o_value = { {(5){i_value[3]}}, i_value[2:0] };``

Actually, `k` is 4 after the for-loop is completed, not 3. Updated test case with wider `i_value`:

```
module mcve1(i_value, o_value);
	input	wire	[4:0]	i_value;
	output	reg	[7:0]	o_value;

	integer k;
	always @(*)
	begin
		for(k=0; k<4; k=k+1)
			o_value[k] = i_value[k];
		o_value[4] = i_value[k];
		o_value[5] = i_value[k];
		o_value[6] = i_value[k];
		o_value[7] = i_value[k];
	end
endmodule
```

And this one is actually broken. Fixed in #972.

(Verific results are looking fine with the updated test case.)

ZipCPU, 2019-04-30:
I just tested this patch within a much larger design, and so I'm now getting driver-driver conflicts based upon the loop variables.

cliffordwolf, 2019-04-30:
@ZipCPU can you send me that design?

ZipCPU, 2019-04-30:
You should have a copy now.

With this patch, yosys also generates the same warnings when verifying [this design](https://github.com/ZipCPU/wb2axip/blob/master/rtl/wbxbar.v).  See [this sby file for reference](https://github.com/ZipCPU/wb2axip/blob/master/bench/formal/wbxbar.v)

cliffordwolf, 2019-04-30:
Disabled the change in e35fe13. Here is an MCVE that even segfaults:

```
module test (input clk, output [3:0] foo);
	integer k;
	always @(posedge clk) begin
		for (k=0; k<5; k=k+1) begin end
	end
	always @(posedge clk) begin
		for (k=0; k<10; k=k+1) begin end
	end
	assign foo = k;
endmodule
```

I have no idea what value should be assigned to `foo` in this case.

cliffordwolf, 2019-04-30:
Segfault fixed in 32ff37b.

cliffordwolf, 2019-04-30:
This is the proper MCVE:

```
module test (input clk, a, b, output reg x, y);
	integer k;
	always @(posedge clk) begin
		for (k=0; k<2; k=k+1) begin x <= a; end
	end
	always @* begin
		for (k=0; k<5; k=k+1) begin y = b; end
	end
endmodule
```

I'm not sure yet what the correct solution is for this. I mean, _technically_ there is a driver driver conflict here for k..

cliffordwolf, 2019-04-30:
Two little test cases:

```
module test1 (input clk, a, b, output reg [3:0] p, q, x, y);
        integer k;
        always @(posedge clk) begin
                for (k=0; k<2; k=k+1)
                        p[2*k +: 2] = {a, b} ^ {2{k}};
                x <= k + {a, b};
        end
        always @* begin
                for (k=0; k<4; k=k+1)
                        q[k] = {~a, ~b, a, b} >> k[1:0];
                y = k - {a, b};
        end
endmodule

module test2 (input clk, a, b, output reg [3:0] q, x, output [3:0] y);
        integer k;
        always @* begin
                for (k=0; k<4; k=k+1)
                        q[k] = {~a, ~b, a, b} >> k[1:0];
        end
        always @* begin
                x = k + {a, b};
        end
        assign y = k - {a, b};
endmodule
```

With 9af825e we get then both right afaict, but we get ``Warning: Driver-driver conflict for \k [0] between cell $procdff$43.Q and constant 1'0 in test1: Resolved using constant.`` warnings.

Without it we get them both wrong.

cliffordwolf, 2019-05-01:
I think/hope #975 provides a proper fix. Please test.

ZipCPU, 2019-05-02:
Nope, this isn't there yet.  I'm still getting the warnings.  These include my [WB crossbar](https://github.com/ZipCPU/wb2axip/blob/master/rtl/wbxbar.v) referenced above, but also my [AXI-lite crossbar](https://github.com/ZipCPU/wb2axip/blob/master/rtl/axilxbar.v) as well.

cliffordwolf, 2019-05-03:
@ZipCPU Just pushed some updates to #975. Please test again. I don't see any driver-driver conflict warnings now for `wbxbar.v` or `axilxbar.v`.

cliffordwolf, 2019-05-06:
@ZipCPU Do you still see driver-driver conflict warnings that shouldn't be there in HEAD of https://github.com/YosysHQ/yosys/tree/clifford/fix968? Can I merge #975?

cliffordwolf, 2019-05-06:
I've now merged #975. Let me know if there are any issues. Closing for now.

closed 2019-05-06 by cliffordwolf

ZipCPU, 2019-05-07:
Sorry for not getting back to you on this sooner.  I have tested the patch and no longer get any driver-driver conflict warnings.

commit 9af825e3 (2019-04-30): Add final loop variable assignment when unrolling for-loops, fixes #968