YosysHQ/yosys #981: Infinite loop while synthesizing for iCE40

ZirconiumX, 2019-05-02:
## Steps to reproduce the issue

Using [attacks.v](https://github.com/YosysHQ/yosys/files/3138569/attacks.zip) with a version of Yosys >= commit e158ea20, run `yosys -p "read_verilog attacks/attacks.v; synth_ice40"`.

## Expected behavior

The program terminates, synthesising the Verilog for an iCE40 and outputting the number of cells used.

## Actual behavior

Yosys loops with repeating output like this:
```
2.27.115. Executing OPT_MUXTREE pass (detect dead branches in mux trees).
Running muxtree optimizer on module \top..
  Creating internal representation of mux trees.
  No muxes found in this module.
Removed 0 multiplexer ports.

2.27.116. Executing OPT_REDUCE pass (consolidate $*mux and $reduce_* inputs).
  Optimizing cells in module \top.
Performed a total of 0 changes.

2.27.117. Executing OPT_MERGE pass (detect identical cells).
Finding identical cells in module `\top'.
Removed a total of 0 cells.

2.27.118. Executing OPT_RMDFF pass (remove dff with constant values).

2.27.119. Executing OPT_CLEAN pass (remove unused cells and wires).
Finding unused cells or wires in module \top..
Removed 98 unused cells and 2988 unused wires.

2.27.120. Executing OPT_EXPR pass (perform const folding).
Optimizing module top.
<suppressed ~376 debug messages>

2.27.121. Rerunning OPT passes. (Maybe there is more to do..)
```
where it seems OPT_CLEAN does not actually remove anything.

*Please describe how the behavior you see differs from the expected behavior.*

I don't know how to reduce the (generated from nMigen) Verilog since I am new to HDL, but I have bisected and verified that this behaviour does not happen on commits before e158ea20.

cliffordwolf, 2019-05-03:
Thanks for reporting this issue. 4219020 / #985 should fix this. Please test and verify.

ZirconiumX, 2019-05-03:
Indeed it does fix the issue, thank you very much!

closed 2019-05-03 by ZirconiumX

ZirconiumX, 2019-05-03:
For what it's worth, I release that Verilog into the public domain, so you can use it or an MCVE based on it as a regression test.

PR #985 (2019-05-03): Improve opt_expr and opt_clean handling of (partially) undriven and/or unused wires
This should fix #981
commit 42190207 (2019-05-03): Improve opt_expr and opt_clean handling of (partially) undriven and/or unused wires, fixes #981