YosysHQ/yosys #982: INIT attribute not set on FDRE when shift register inference

eddiehung, 2019-05-02:
## Steps to reproduce the issue

`yosys -p "synth_xilinx; dump t:FDRE" lfsr_102.v`

```
module lfsr_102 (input clk, output dout);
  reg [102:1] state = 102'b0;
  always @(posedge clk)
    state <= { state[102-1:1], state[102] ~^ state[101] ~^ state[36] ~^ state[35] };
  assign dout = state[102];
endmodule
```

## Expected behavior

`FDRE` to have `INIT` parameter set to `1'b0`

## Actual behavior

`FDRE` has `INIT` parameter set to `1'bx`
```
<<SNIP>>
  cell \FDRE $auto$simplemap.cc:420:simplemap_dff$54.fpga_srl_1
    parameter \INIT 1'x
    connect \C \clk
    connect \CE 1'1
    connect \D $auto$simplemap.cc:420:simplemap_dff$54.T0
    connect \Q \state [100]
    connect \R 1'0
  end
<<SNIP>>
```
Works when shift register inference disabled (`synth_xilinx -nosrl`).
Could this be related to #956 (though there are no warning messages as that issue has)

eddiehung, 2019-05-03:
Issue appears to be that `shregmap` will set away the `(* init *)` attribute on signals that it infers into a shift register to `1'bx`. However, our tech mapping rules can sometimes blow these inferred shift registers back into flops, and when it does that, a subsequent `dffinit` will init those registers at `1'bx`... (even if I have instantiated a tech specific cell (e.g. `FDRE`))

Number of solutions:
- `shregmap` to now blow away `(* init *)` attribute
- `dffinit` to not overwrite cells with `INIT` already set with new `1'bx` value.

Am thinking latter...

cliffordwolf, 2019-05-03:
\o/

```
  cell \FDRE $auto$simplemap.cc:420:simplemap_dff$17.fpga_srl_1
    parameter \INIT 1'0
    connect \C \clk
    connect \CE 1'1
    connect \D $auto$simplemap.cc:420:simplemap_dff$17.T0
    connect \Q \state [16]
    connect \R 1'0
  end
```

closed 2019-05-03 by cliffordwolf