YosysHQ/yosys #987: SystemVerilog var port kind

tilk, 2019-05-03:
The following SystemVerilog program is accepted by Yosys:
```systemverilog
module test(output a, input b);
    always_comb a = b;
endmodule
```
But it is incorrect. IEEE 1800-2012 section 23.2.2.3 says as follows:

> * For output ports, the default port kind depends on how the data type is specified:
>       
>   * If the data type is omitted or declared with the implicit_data_type syntax, the port kind shall default to a net of default net type.
>   * If the data type is declared with the explicit data_type syntax, the port kind shall default to variable.

The correct version of the program is:
```systemverilog
module test(output var a, input b);
    always_comb a = b;
endmodule
```
But it is not parsed correctly. I get:
```
ERROR: syntax error, unexpected TOK_ID, expecting ',' or '=' or ')'
```

cliffordwolf, 2019-05-04:
> The following SystemVerilog program is accepted by Yosys: [..] But it is incorrect.

That's why you get ``Warning: wire '\a' is assigned in a block at test.sv:2.`` when parsing it with `read_verilog`.

> The correct version of the program is

#988 adds hacky support for the SV `var` keyword, Please test.

tilk, 2019-05-04:
Tested, works for me.

PR #988 (2019-05-04): Add approximate support for SV "var" keyword
Fix #987
commit 9804c86e (2019-05-04): Add approximate support for SV "var" keyword, fixes #987
closed 2019-05-04 by cliffordwolf