YosysHQ/yosys #994: Some warning fixes.

Kmanfi, 2019-05-06:
## Steps to reproduce the issue

frontends/ilang/ilang_parser.y:48.1-38: warning: deprecated directive, use ‘%define api.prefix {rtlil_frontend_ilang_yy}’ [-Wdeprecated]
 **%name-prefix "rtlil_frontend_ilang_yy"**
Fix:
**%define api.prefix {rtlil_frontend_ilang_yy}**
__
frontends/verilog/verilog_parser.y:99.1-34: warning: deprecated directive, use ‘%define api.prefix {frontend_verilog_yy}’ [-Wdeprecated]
 **%name-prefix "frontend_verilog_yy"**
Fix:
**%define api.prefix {frontend_verilog_yy}**

## Expected behavior

No warning.
## Actual behavior

cliffordwolf, 2019-05-06:
I think that was fixed by #992 approx 2 hours ago.

cliffordwolf, 2019-05-06:
Fixed the `ilang_parser.y` warning in 20268d1.

closed 2019-05-06 by Kmanfi