YosysHQ/yosys #995: Python bindings, OS X

Kmanfi, 2019-05-06:
## Steps to reproduce the issue
FYI.
To get Python bindings working on **my** OS X machine I made following modifications to Makefile.

ENABLE_LIBYOSYS := 0
ENABLE_LIBYOSYS := **1**
__
PYTHON_DESTDIR := `$(PYTHON_EXECUTABLE)-config --prefix`/lib/python$(PYTHON_VERSION)/dist-packages

PYTHON_DESTDIR := `$(PYTHON_EXECUTABLE)-config --prefix`/lib/python$(PYTHON_VERSION)/**site**-packages
__
ENABLE_NDEBUG := 1
ENABLE_NDEBUG := **0**
__
ifneq ($(shell which brew),)
BREW_PREFIX := $(shell brew --prefix)/opt
$(info $$BREW_PREFIX is [${BREW_PREFIX}])
**CXXFLAGS += -I$(BREW_PREFIX)/boost/include/boost
LDFLAGS += -L$(BREW_PREFIX)/boost/lib**
__
All..
$(PYTHON_EXECUTABLE)-config --libs
$(PYTHON_EXECUTABLE)-config --**ldflags**
__
$(P) $(LD) -o libyosys.so -shared -Wl,-soname,libyosys.so $(LDFLAGS) $^ $(LDLIBS)
$(P) $(LD) -o libyosys.so -shared -Wl,-install_name,libyosys.so $(LDFLAGS) $^ $(LDLIBS)
__
$(INSTALL_SUDO) ldconfig
**#** $(INSTALL_SUDO) ldconfig

## Expected behavior

Python bindings work also on OS X.
## Actual behavior

wip

cliffordwolf, 2019-05-06:
CC @btut @christian-krieg

Kmanfi, 2019-05-07:
If I only change
  ENABLE_NDEBUG := 1
then
I get many of these:

```
In file included from kernel/rtlil.cc:20:
In file included from ./kernel/yosys.h:270:
./kernel/log.h:93:33: warning: unused parameter 'n' [-Wunused-parameter]
static inline bool ys_debug(int n = 0) { return false; }
```

btut, 2019-05-07:
Hi @Kmanfi,
Most of the changes you applied to the makefile can be safely applied without breaking my linux build. I guess it would be safe to apply those to the repo.

This is not true for the following line:
`$(P) $(LD) -o libyosys.so -shared -Wl,-install_name,libyosys.so $(LDFLAGS) $^ $(LDLIBS)`
-install_name is a darwin specific flag. A check is needed here to make sure this is only changed for darwin. I am not completely sure though, as that is not part of the python wrapper build, please confirm @cliffordwolf 

For the NDEBUG issue, this does not seem to be related to python. Can you build with:
```
ENABLE_NDEBUG := 1
ENABLE_PYOSYS := 0
```
?

I am a little unsure how to interpret the actual behavior you are describing. You are providing fixes to the makefile, are those not sufficient to build yosys with python wrappers? If they are, are you running into any run-time issues?

Kmanfi, 2019-05-08:
Hello,
does this work on your machine?
```
for module in design.selected_whole_modules_warn():
    for conn in module.connections_:
            doo = 0

```

Kmanfi, 2019-05-09:
Found a bug related to the previous example. Needs fix in py_wrap_generator.py
`#Generate c++ code to translate to a boost::python::tuple
...`
See hand made fix to generated file that works.
```
	boost::python::list Module::get_var_py_connections_()
	{
		std::vector<std::pair<YOSYS_NAMESPACE::RTLIL::SigSpec, YOSYS_NAMESPACE::RTLIL::SigSpec>> ret_ = this->get_cpp_obj()->connections_;
		boost::python::list ret____tmp;
		for(auto tmp_116 : ret_)
		{
			// currently generated NOT working boost::python::tuple tmp_116___tmp = boost::python::make_tuple(tmp_116.first, tmp_116.second);
			boost::python::tuple tmp_116___tmp = boost::python::make_tuple(*SigSpec::get_py_obj(&tmp_116.first), *SigSpec::get_py_obj(&tmp_116.second));
			ret____tmp.append(tmp_116___tmp);
		}
		return ret____tmp;
	}

```

Kmanfi, 2019-05-09:
"I am a little unsure how to interpret the actual behavior you are describing. You are providing fixes to the makefile, are those not sufficient to build yosys with python wrappers? If they are, are you running into any run-time issues?"

Current state:
- python wrappers compile
- script.py works
- pass.py exit with ``ERROR: Assert `first_queued_pass == NULL' failed in kernel/register.cc:69.``

Kmanfi, 2019-05-09:
This is my test file.
```
module counter (clk, aload, d, b, q);
input clk, aload;
input  [3:0] d;
input  [3:0] b;
output [7:0] q;
reg    [3:0] tmp;

  always @(posedge clk)
    begin
      if (aload)
        tmp = d + b;
      else
        tmp = tmp + 1'b1;
    end
  assign q[3:0] = tmp;
  assign q[4] = 1;
  assign q[5] = d[2];
  assign q[6:7] = tmp[1:0];
endmodule
```
And this is current "test" that fail to 
RuntimeError: Member "wire" is NULL
```
for module in design.selected_whole_modules_warn():
    for conn_first, conn_second in module.connections_:
        da = sigmap(conn_first).to_sigbit_vector()
        for bit in da:
            print(bit.wire)
```

arcturusannamalai, 2023-05-04:
why not use Make for Yosys ? Seems like a nicer alternative to Makefile IMHO