YosysHQ/yosys #1002: Determining the index of a onehot vector

ZipCPU, 2019-05-10:
## Steps to reproduce the issue

Consider the following module, whose purpose is to output the index of the one-hot vector value given to it.

```
module onehot(i_in, o_out);
	parameter	LG = 7;
	localparam	WID = (1<<LG);
	//
	input	wire	[WID-1:0]	i_in;
	output	reg	[LG-1:0]	o_out;

	//
	integer	N;

	always @(*)
	begin
		o_out = 0;
		for(N=0; N<WID; N=N+1)
		begin
			if (i_in[N])
				o_out = o_out | N[LG-1:0];
		end
	end
endmodule
```

This module can be synthesized and displayed with Yosys using:

```
yosys -p 'read -sv onehot.v ; hierarchy -top onehotsel -chparam LG 3; synth_xilinx; stat; show'
```

## Expected behavior

For LG = 3 and LG = 4, Yosys has no problems: the design synthesizes fine, and you can see and examine the logic via the show command.

## Actual behavior

If LG is anything larger than 4, Yosys sets the output vector to all ones, discarding all of the logic within.  Given that LG=5, the first failing value, hits the 32-bit wordsize my guess is that these two are somehow related.

The SymbioticEDA Suite, version 20190416A, has no problems parsing or synthesizing this design, so this seem specific to the open source version.

cliffordwolf, 2019-05-11:
Seems to be an issue in `opt_muxtree`. I'm investigating..

cliffordwolf, 2019-05-12:
Should be fixed by #1004. I'll merge it as soon as Travis goes green.

PR #1004 (2019-05-12): Fix handling of glob_abort_cnt in opt_muxtree
Fixes #1002
commit 8166a142 (2019-05-12): Fix handling of glob_abort_cnt in opt_muxtree, fixes #1002
closed 2019-05-12 by cliffordwolf