YosysHQ/yosys #1006: CelleEdgesDatabase needs fix for shift operations

aJunk, 2019-05-13:
We are working on a project using a CellEdgesDatabase to create a data-flow graph.
I have notices, that shift operations are not mapped an there is a 'fix me' comment in 'kernel/celledges.cc'.

If there anything going against treating all input bits as influencing all output bits?

I have patched my local install like I mentioned above and it works properly for the algorithm we are implementing.
If that makes sense I will gladly submit a pull request. 

I am sorry if this is in the wrong place. I have not much experience in trying to contribute to such a large project.

eddiehung, 2019-05-15:
Hi @aJunk! Thanks for contributing to Yosys! Please create a PR with your proposed changes, along with a test to demonstrate its effectiveness (and to make sure it never gets broken in the future). The relevant people will then review it, offer changes if necessary, and merge it in. I'd be happy to help you along if you get stuck.

aJunk, 2019-05-15:
Hi, thank you for your reply!

I will create a PR then. If I may I will send you an eMail regarding how to implement the testing part if I can not figure it out by myself.

Thank you again for your help :)