YosysHQ/yosys #1009: Attributes on parameters / localparams are not parsed correctly per Verilog 2005 spec

litghost, 2019-05-14:
## Steps to reproduce the issue

1. Run "yosys [example.v](https://gist.github.com/litghost/2f4201d0983c0d2736de5c2cd5e47e72)"

## Expected behavior

Yosys parses the file per the [Verilog 2005 spec](http://staff.ustc.edu.cn/~songch/download/IEEE.1364-2005.pdf)

Per:
```
module_or_generate_item ::=
 { attribute_instance } module_or_generate_item_declaration
| { attribute_instance } local_parameter_declaration ;
| { attribute_instance } parameter_override
| { attribute_instance } continuous_assign
| { attribute_instance } gate_instantiation
| { attribute_instance } udp_instantiation
| { attribute_instance } module_instantiation
| { attribute_instance } initial_construct
| { attribute_instance } always_construct
| { attribute_instance } loop_generate_construct
| { attribute_instance } conditional_generate_construct
non_port_module_item ::=
 module_or_generate_item
| generate_region
| specify_block
| { attribute_instance } parameter_declaration ;
| { attribute_instance } specparam_declaration
```

The Verilog 2001 specification also indicates that attributes are supported on parameter declarations:

```
non_port_module_item ::=
 { attribute_instance } generated_instantiation
| { attribute_instance } local_parameter_declaration
| { attribute_instance } module_or_generate_item
| { attribute_instance } parameter_declaration
| { attribute_instance } specify_block
| { attribute_instance } specparam_declaration 
```

## Actual behavior

```
1. Executing Verilog-2005 frontend.
Parsing Verilog input from `example.v' to AST representation.
example.v:3: ERROR: syntax error, unexpected TOK_PARAMETER
```

mkurc-ant, 2019-05-14:
I'd like to work on that issue as it is very relevant to what I am doing. I need support for parameters with attributes.

mkurc-ant, 2019-05-15:
I think that this issue depends to #1001.

cliffordwolf, 2019-05-15:
> I need support for parameters with attributes.

Do you just need the parser to accept the parameter with attribute syntax, or do you actually want to see the parameter in some way in the RTLIL output? Because the former is easy but the latter might be a bit of a challenge..

mkurc-ant, 2019-05-15:
Hi, I need the latter. I am using Yosys to convert Verilog to JSON representation. And I need to have both parameters on modules and their attributes exported to JSON.

I started to work my way through the Yosys code and I think I have a solution: I propose to add new class `RTLIL::Parameter` and have `dict<RTLIL::IdString, RTLIL::Parameter*> parameters_` inside the `RTLIL::Module` struct. It would work just like with eg. `RTLIL::Wire`.

Moreover I managed to make the Verilog parser accept attributes on parameters but I do not know yet how to attach them to the `AstNode` object.

What do you think? I have some WIP code that I can push.

cliffordwolf, 2019-05-15:
> I started to work my way through the Yosys code and I think I have a solution: I propose to add new class RTLIL::Parameter and have dict<RTLIL::IdString, RTLIL::Parameter*> parameters_ inside the RTLIL::Module struct. It would work just like with eg. RTLIL::Wire.
> ...
> What do you think? I have some WIP code that I can push.

Hmm... Not without a clearer motivation how and why this is useful.

Yosys output is after elaboration. At this time we should be done with parameters. So why do we need them at this stage?

mkurc-ant, 2019-05-15:
So my motivation is to have an attribute list associated with each module parameter. And currently I see that the `RTLIL::Module` has only a list of its parameters (`pool<RTLIL::IdString> avail_parameters`). No place to store attributes.

So I want to effectively replace that list with `dict<RTLIL::IdString, RTLIL::Parameter*>`. And the new object type `RTLIL::Parameter` will be capable of holding attribute information plus the default value of the parameter (as also requested in #1001).

The elaboration should not affect attributes of parameters.

litghost, 2019-05-15:
> > I need support for parameters with attributes.
> 
> Do you just need the parser to accept the parameter with attribute syntax, or do you actually want to see the parameter in some way in the RTLIL output? Because the former is easy but the latter might be a bit of a challenge..

I think a good starting point would be for Yosys to be able to round trip legal Verilog 2005 from the Verilog frontend to Verilog backend preserving the attributes on parameters.  Other backends (specifically JSON) are missing some information pertaining to parameters per #1001, so as part of the #1001 work, output to additional backends could be added.

cliffordwolf, 2019-05-15:
> So my motivation is to have an attribute list associated with each module parameter.

I got that. I want to know why you want that. What's your motivation? What do you want to do with this information?

> The elaboration should not affect attributes of parameters.

No. But after elaboration there is no use for parameters anymore. Everything that depends on them has been elaborated. And thus I'd like to know what you are hoping to gain from preserving the parameters through yosys.

As with all synthesis flows, there is a lot of details in the Verilog input that isn't preserved throughout the flow. Preserving it all would be completely impractical. Usually synthesis tools do not preserve attributes on parameters. Why do you want them preserved?

mkurc-ant, 2019-05-15:
@cliffordwolf We (at Symbiflow) also use the Yosys to convert Verilog description of the FPGA architecture to XML format understandable by the VPR. We have a tool which relies on the Yosys feature which is generating a JSON representation of the Verilog design.

The goal here is not to have a full synthesis flow which breaks up the input design structure to smaller pieces. It is rather to process Verilog input and generate its JSON representation with the module hierarchy and their attributes + parameters.

We use the following Yosys scirpt: `read_verilog my_design.v; prep; write_json my_design.json` and then we process the JSON.

And right now we do not have parameters of the top level module listed in JSON. There are only those of other modules instances (#1001). And also we need attributes on parameters. We plan to use them for metadata storage (this issue - #1009).

eddiehung, 2019-05-15:
I agree with @cliffordwolf here. Yosys is not a source to source translator (i.e. round trip Verilog to Verilog preservation) but a synthesis (elaboration) tool. Thus, this feature is one that does not provide any benefits for synthesis, and the code and memory overhead cost for your use case case should not be bourne by all other users of Yosys that have no use for this feature. This is an ideal case for keeping this feature in a separate fork and not upstreaming.

I'll give some more insight: Verific's commercial elaborator (those often used by FPGA vendors) do not preserve parameters during elaboration either.

mkurc-ant, 2019-05-16:
Ok, if you insist we can focus on just making the Yosys' Verilog frontend conform to Verilog 2001 and 2005 spec. This requires that it can parse code with attributes on parameters. Attribute content may be discarded. And the behavior regarding parameter preservation remains unchanged.

closed 2019-05-16 by cliffordwolf

mithro, 2019-05-16:
@cliffordwolf / @eddiehung - Just to be clear we are not asking for a major change in how Yosys behaves. Currently if you have a verilog file like;
```verilog
module ABC(i, o);
  input wire i;
  output wire o;
  parameter p = 0;
  assign o = i;
endmodule

module top(i, o);
  input wire i;
  output wire o;

  wire m;
  ABC #(.p(1)) a1(.i(i), .o(m));
  ABC #(.p(2)) a2(.i(m), .o(o));
endmodule
```

```
yosys> read_verilog "/home/tansell/temp.v"
1. Executing Verilog-2005 frontend: /home/tansell/temp.v
Parsing Verilog input from `/home/tansell/temp.v' to AST representation.
Generating RTLIL representation for module `\ABC'.
Generating RTLIL representation for module `\top'.
Successfully finished Verilog frontend.

yosys> ls

2 modules:
  ABC
  top

yosys> chparam -list
top:
ABC:
  p

yosys> cd ABC

yosys [ABC]> chparam -list
ABC:
  p

yosys [ABC]> dump

autoidx 1

attribute \cells_not_processed 1
attribute \src "/home/tansell/temp.v:1"
module \ABC

  parameter \p

  attribute \src "/home/tansell/temp.v:3"
  wire output 2 \o

  attribute \src "/home/tansell/temp.v:2"
  wire input 1 \i

  connect \o \i
end
```

The only change we would expect is that if you have;
```verilog
module ABC(i, o);
  input wire i;
  output wire o;

  (* hello *)
  parameter p = 0;
  assign o = i;
endmodule

module top(i, o);
  input wire i;
  output wire o;

  wire m;
  ABC #(.p(1)) a1(.i(i), .o(m));
  ABC #(.p(2)) a2(.i(m), .o(o));
endmodule
```

Then the `dump` output would change as follows;
```
yosys [ABC]> dump

autoidx 1

attribute \cells_not_processed 1
attribute \src "/home/tansell/temp.v:1"
module \ABC

  attribute \hello 1
  parameter \p

  attribute \src "/home/tansell/temp.v:3"
  wire output 2 \o

  attribute \src "/home/tansell/temp.v:2"
  wire input 1 \i

  connect \o \i
end
```

We can then use the new Python pass API to access the RTIL parameter object and see the attribute values on parameters and manipulates the RTIL using things like `setparam` and `chparam`.

A current "work around" is to add the attribute (with a special name that matches the parameter) to the top level object (rather than on the parameter itself). This seems like a nasty hack when it seems like a pretty simple addition to Yosys?

```verilog
(* p_hello *)
module ABC(i, o);
  input wire i;
  output wire o;

  parameter p = 0;
  assign o = i;
endmodule
```

This also has the added benefit that parameters could also get `src` attributes like everything else enabling things like the parameter functions to list line numbers where the parameter was defined which could improve help and error messages.

It is my understanding that as this is a change which happens to the parser & RTIL we can't implement this in the Python / loadable module?

PR #1013 (2019-05-16): Support for attributes on parameters and localparams for Verilog frontend
This PR adds to the Verilog frontend support for attributes on the "parameter" and "localparam" statements.  Attributes are just accepted by the parser and their content is ignored. There are also tests added regarding this functionality (under `tests/simple`)

Fixes #1009