YosysHQ/yosys #1014: Generate loop hang

Kmanfi, 2019-05-16:
## Steps to reproduce the issue
```
module top (input [7:0] data, output [7:0] out);
genvar n;
generate
    for (n=7; n>=0; n=n-1) begin
        assign out[n] = data[7-n];
    end
endgenerate
endmodule
```

## Expected behavior

Connect eight wires in a seconds. 😁 
## Actual behavior

Run out of memory. 😅

daveshah1, 2019-05-16:
As far as I know, `genvar`s are non negative integers (what other tools treat them as I don't know) so this isn't strictly speaking valid code - although hanging and running out of memory is by no means the best way to behave here!

udif, 2019-05-16:
From IEEE1800-2017, section 27.4:

> The genvar is used as an **integer** during elaboration

As far as I can see, nothing here requires genvars to be unsigned.
Running the test code above works fine on both Verilator and Cadence Incisive simulator.
(I was too lazy to try on our synthesis tools, as it requires a tedious setup script).

udif, 2019-05-16:
While reading the relevant section on the LRM, I was quite surprised to learn that using the generate/endgenerate keyword pair is purely optional!
The code below
```
module top (input [7:0] data, output [7:0] out);
genvar n;
    for (n=0; n <= 7; n=n+1) begin
        assign out[n] = data[7-n];
    end
endmodule
```
Works as-is on Yosys, Verilator, and Cadence Incisive.

daveshah1, 2019-05-16:
Maybe this has changes between Verilog versions? 1364-2001 states

 >  If any bit of the genvar ever is set to an X or Z or if the genvar
is set to a negative value, this shall be an error.

cliffordwolf, 2019-05-16:
> While reading the relevant section on the LRM, I was quite surprised to learn that using the generate/endgenerate keyword pair is purely optional!

That's a SystemVerilog feature. In Verilog the generate keyword is mandatory.

cliffordwolf, 2019-05-16:
> Maybe this has changes between Verilog versions? 1364-2001 states

Looks like this is the case. Page 183 of 1364-2005:

[image]

Apparently they removed the clause regarding negative values.

udif, 2019-05-16:
@cliffordwolf , you're quicker than me.
Yes, I also found they dropped it in Verilog-AMS, IEC 61691-4, and the 1364-2004 drafts...

cliffordwolf, 2019-09-19:
This seems to be fixed in current Yosys master. Thus I am closing this issue.

Kmanfi, 2019-09-23:
Tested with original code, not @udif version.
Yosys 0.9+431 (git sha1 40cd3841, clang 10.0.1 -fPIC -Os)
Bug still exists. Attached is better test case.
```
module top (input [7:0] data, output [7:0] out);
genvar n;
generate
    for (n=7-1; n>=0-1; n=n-1) begin
        assign out[n+1] = data[7-n+1];
    end
endgenerate
endmodule
```

koriakin, 2019-09-24:
I have no idea where you got your yosys from (I cannot see commit 40cd3841 anywhere in history), but your example works on current master (057dae4f78ef604964fb12175f5b263bb386da24).

cliffordwolf, 2019-09-25:
@Kmanfi I can not see any problem with yosys `read_verilog` with this code in current master (739c621). Like @koriakin I cannot see a commit 40cd3841.

There is however what looks like a bug in your code. I would assume you meant to write `data[7-(n+1)]`, or `data[7-n-1]`. (Yosys processes your input correctly and produces the circuit it describes, but presumably that isn't the circuit you wanted.)

Kmanfi, 2019-09-26:
Cloned master to clean directory. All worked as expected. Thanks.

This works after I fixed bug @cliffordwolf noted.
```
module top (input [7:0] data, output [7:0] out);
genvar n;
generate
    for (n=7-1; n>=0-1; n=n-1) begin
        assign out[n+1] = data[7-(n+1)];
    end
endgenerate
endmodule

```

closed 2019-09-26 by Kmanfi