YosysHQ/yosys #1015: Output cone select appears to no longer work on blackboxes?

litghost, 2019-05-17:
## Steps to reproduce the issue

1. Download [stub.v](https://gist.github.com/litghost/83ee196a7eb529138fba5bf43376822a)
2. Run:
```
yosys -p "cd STUB; select IN %co*; ls" stub.v
```

## Expected behavior

Select statement should report connected wires from IN wires (which worked at rev: f84a84e3f1a27b361c21fcd30fcf50c1a6586629 )

```
Yosys 0.8+332 (git sha1 fb7f02be, clang 4.0.1-10 -fPIC -Os)

-- Parsing `stub.v' using frontend `verilog' --

1. Executing Verilog-2005 frontend.
Parsing Verilog input from `stub.v' to AST representation.
Generating RTLIL representation for module `\STUB'.
Successfully finished Verilog frontend.

-- Running command `cd STUB; select IN %co*; ls' --

2 wires:
  IN
  OUT

End of script. Logfile hash: 9863935eae
CPU: user 0.01s system 0.00s, MEM: 40.72 MB total, 6.68 MB resident
Yosys 0.8+331 (git sha1 f84a84e3, clang 4.0.1-10 -fPIC -Os)
Time spent: 49% 1x select (0 sec), 44% 1x read_verilog (0 sec), ...
```

## Actual behavior

Select statement no longer finds connected wires (as of fb7f02be5561ccfd5bee5f3235fbbae5ef618f36 )

```
 Yosys 0.8+332 (git sha1 fb7f02be, clang 4.0.1-10 -fPIC -Os)

-- Parsing `stub.v' using frontend `verilog' --

1. Executing Verilog-2005 frontend.
Parsing Verilog input from `stub.v' to AST representation.
Generating RTLIL representation for module `\STUB'.
Successfully finished Verilog frontend.

-- Running command `cd STUB; select IN %co*; ls' --

1 wires:
  IN

End of script. Logfile hash: 53919bc75e
CPU: user 0.00s system 0.00s, MEM: 40.73 MB total, 6.60 MB resident
Yosys 0.8+332 (git sha1 fb7f02be, clang 4.0.1-10 -fPIC -Os)
Time spent: 51% 1x select (0 sec), 42% 1x read_verilog (0 sec), ...
```

daveshah1, 2019-05-18:
The Yosys README says:
> The blackbox attribute on modules is used to mark empty stub modules that have the same ports as the real thing but do not contain information on the internal configuration. This modules are only used by the synthesis passes to identify input and output ports of cells.

As far as I know, it was a bug that previously the content of blackbox modules was being preserved and used for identifying cones. 

The new whitebox attribute is intended for this situation, where you want a module to behave like a blackbox for synthesis purposes (e.g. by remaining as a cell instance rather than being flattened); but its content to be available for purposes such as identifying cones, or in the future more advanced synthesis optimisations.

> The whitebox attribute on modules triggers the same behavior as blackbox, but is for whitebox modules, i.e. library modules that contain a behavioral model of the cell type.

Chaning `(* blackbox *)` to `(* whitebox *)` and the testcase works properly.

cliffordwolf, 2019-05-18:
What @daveshah1 said. We are now explicitly removing all logic from blackbox modules. That includes direct connections from inputs to outputs.

closed 2019-05-18 by cliffordwolf