YosysHQ/yosys #1018: Possible regression in b0b2be8

adumont, 2019-05-19:
My project () build automatically the toolchain in Travis CI, and then it synthesize the bitstream. I haven't changed anything in the HDL, but this week the build has failed. Last week it was OK.

Some info:

SUCCESS build #108 - LOG: https://travis-ci.org/adumont/hrm-cpu/builds/531446647

Toolchain commits used in build 108:
/home/travis/toolchain/arachne-pnr.ver:840bdfdeb38809f9f6af4d89dd7b22959b176fdd
/home/travis/toolchain/icestorm.ver:dd8c003e796d570fc93d35e18dc0cce820c46021
/home/travis/toolchain/symbiyosys.ver:f087a71f499f77970e43805fa48df7465aeeed6c
/home/travis/toolchain/yices2.ver:6dc67e9d1c11e22c40ea5ac110ead56acb9e2195
/home/travis/toolchain/yosys.ver:5772732a6d24bbf1a46301f03623567079316f7a

FAILED build #109 - LOG: https://travis-ci.org/adumont/hrm-cpu/builds/534482459

Toolchain commits used in build 109:
/home/travis/toolchain/arachne-pnr.ver:840bdfdeb38809f9f6af4d89dd7b22959b176fdd <-- same
/home/travis/toolchain/icestorm.ver:d78e9185f86293442b361990a7a2b3f5bf30c173   <-- changed
/home/travis/toolchain/symbiyosys.ver:f087a71f499f77970e43805fa48df7465aeeed6c <-- same
/home/travis/toolchain/yices2.ver:4328f5b43e0d73bf58008d78b8e2f771d27f6358  <-- changed (not related)
/home/travis/toolchain/yosys.ver:c907899422884d959632ed42c6589a0720b681e4  <-- changed

Error (best seen in the build log, see link above)
arachne-pnr -d 1k -p top-alhambra.pcf builddir/alhambra/top.blif -o builddir/alhambra/top.pnr
[...]
fatal error: failed to route

I've done a git bisect run,  rebuilding Yosys and my project and it gave me this log:

git bisect start
# good: [5772732a6d24bbf1a46301f03623567079316f7a] Merge pull request #1004 from YosysHQ/clifford/fix1002
git bisect good 5772732a6d24bbf1a46301f03623567079316f7a
# bad: [c907899422884d959632ed42c6589a0720b681e4] Merge pull request #1017 from Kmanfi/bigger_verilog_files
git bisect bad c907899422884d959632ed42c6589a0720b681e4
# good: [64b604207deefb87faa1aaf158012b7063526c6f] Merge pull request #1010 from hzeller/yacc-self-contained
git bisect good 64b604207deefb87faa1aaf158012b7063526c6f
# bad: [c9def5407cd59e8542201217c890bbe4c89fff5c] Merge pull request #1012 from YosysHQ/clifford/sigspecrw
git bisect bad c9def5407cd59e8542201217c890bbe4c89fff5c
# good: [f67ec1b2350c3bd88edacba7ad799bf60b33da61] Do not leak file descriptors in cover.cc
git bisect good f67ec1b2350c3bd88edacba7ad799bf60b33da61
# bad: [a21a84b3b45d252ff050d3ae7e738d157b0b2be8] Improvements in opt_clean
git bisect bad a21a84b3b45d252ff050d3ae7e738d157b0b2be8
# good: [287de4b848bdd33a9ba52018fc9b9779a10e60c7] Add rewrite_sigspecs2, Improve remove() wires
git bisect good 287de4b848bdd33a9ba52018fc9b9779a10e60c7
# first bad commit: [a21a84b3b45d252ff050d3ae7e738d157b0b2be8] Improvements in opt_clean

Hope that helps.

adumont, 2019-05-19:
I attach two files: output of yosys both in "good" and "bad" commit. 

In both case, this blf file was generated with:

yosys -p "synth_ice40 -top top -blif builddir/alhambra/top.blif " \
              -l builddir/alhambra/BUMP+/top.log -q hrmcpu.v ufifo.v ALU.v MEMORY.v ram.v REG.v IR.v PROG.v PC.v ControlUnit.v debouncer.v rxuartlite.v txuartlite.v builddir/alhambra/top_wrapper.v top.v

In the bad commit, it fails to PNR with arachne-pnr (see https://travis-ci.org/adumont/hrm-cpu/builds/534482459#L4790):

arachne-pnr -d 1k -p top-alhambra.pcf builddir/alhambra/top.blif -o builddir/alhambra/top.pnr

[top.blif.bad.txt](https://github.com/YosysHQ/yosys/files/3195667/top.blif.bad.txt)
[top.blif.good.txt](https://github.com/YosysHQ/yosys/files/3195668/top.blif.good.txt)

daveshah1, 2019-05-20:
This is a known problem in arachne-pnr. Occasionally it can place the design in a way that is impossible to route, even if a routeable placement should easily be achievable. Place and route algorithms are highly dependent on the connectivity and ordering of cells - this bug could be triggered by changing the seed value in arachne-pnr without changing the Yosys output at all. Any small change in Yosys could have triggered this issue.

I would suggest looking at nextpnr to see if it works for you, as it does not have this bug and tends to produce better quality results than arachne-pnr.

adumont, 2019-05-20:
OK. Travis CI automatically rebuild this same RTL design every week
(rebuilding the latest version of the toolchain, as you can see in the
builds history logs), and it always build fine, unless there a regression
in the toolchain (unless I mess up my RTL, which I haven't changed in
weeks).

So I was thinking maybe this would be the case here as well.

I will looking into nextpnr to fix the issue.

Dobyou think I could also try another seed? Maybe a new seed could help
with the PNR with arachnepnr?

daveshah1, 2019-05-20:
Using -s seed or -r for a random seed will almost certainly get the design routing again.

cliffordwolf, 2019-05-20:
Not a Yosys issue. Closing.

closed 2019-05-20 by cliffordwolf

adumont, 2019-05-20:
Just to follow up, tested with a random seed and arachne-pnr now passes fine. Thanks.
https://travis-ci.org/adumont/hrm-cpu/builds/534754564