YosysHQ/yosys #1022: Error parsing systemverilog integer literal constants

AaronKel, 2019-05-21:
## Expected behavior

I expect the following code

```
module test(
  output logic [31:0] b
);

  assign b = '0;

endmodule
```

to be equivalent to

```
module test(
  output logic [31:0] b
);

  assign b = 32'0;

endmodule
```

Looking at the IEEE Std 1800-2017 - 5.7.1 Integer literal constants 

> An unsized single-bit value can be specified by preceding the single-bit value with an apostrophe ( ' ), but without the base specifier. All bits of the unsized value shall be set to the value of the specified bit. In a self-determined context, an unsized single-bit value shall have a width of 1 bit, and the value shall be treated as unsigned.

## Actual behavior

Parsing SystemVerilog input from `test.v' to AST representation.
test.v:6: ERROR: syntax error, unexpected $undefined

AaronKel, 2019-05-21:
This is also the case for '1, 'x and 'z.

sumit0190, 2019-05-24:
I encountered the same problem a couple of days ago and made this half-researched attempt to fix it:

```diff
diff --git a/frontends/verilog/const2ast.cc b/frontends/verilog/const2ast.cc
index 7848c626..13145212 100644
--- a/frontends/verilog/const2ast.cc
+++ b/frontends/verilog/const2ast.cc
@@ -224,7 +224,12 @@ AstNode *VERILOG_FRONTEND::const2ast(std::string code, char case_type, bool warn
 			my_strtobin(data, endptr+2, len_in_bits, 16, case_type);
 			break;
 		default:
-			return NULL;
+            char next_char = char(tolower(*(endptr+1)));
+            if (next_char == '0' || next_char == '1' || next_char == 'x' || next_char == 'X') {
+                my_strtobin(data, endptr+2, -1, 2, case_type);
+            } else {
+	          return NULL;
+            }
 		}
 		if (len_in_bits < 0) {
 			if (is_signed && data.back() == RTLIL::S1)
diff --git a/frontends/verilog/verilog_lexer.l b/frontends/verilog/verilog_lexer.l
index 142d05d4..d89e144a 100644
--- a/frontends/verilog/verilog_lexer.l
+++ b/frontends/verilog/verilog_lexer.l
@@ -232,7 +232,7 @@ YOSYS_NAMESPACE_END
 	return TOK_CONSTVAL;
 }
 
-[0-9]*[ \t]*\'s?[bodhBODH][ \t\r\n]*[0-9a-fA-FzxZX?_]+ {
+[0-9]*[ \t]*\'s?[bodhBODH]*[ \t\r\n]*[0-9a-fA-FzxZX?_]+ {
 	frontend_verilog_yylval.string = new std::string(yytext);
 	return TOK_CONSTVAL;
 }
```
This doesn't work correctly for X and Z though, so I suppose I am clearly on the wrong track here. Any suggestions?

AaronKel, 2019-05-24:
Looks on the right track,
why doesn't my_strtobin() handle the x's? What is returned when you pass it the x's?
Also the standard permits the use of 'z' and 'Z'.

AaronKel, 2019-05-24:
Also you can now parse things like
```
module test(
  output logic [31:0] b
);

  assign b = '00;

endmodule
```
and writing the code out for 
```
module test(
  output logic [31:0] b
);

  assign b = '1;

endmodule
```
is equivalent to 
```
module test(
  output logic [31:0] b
);

  assign b = 32'b1;

endmodule
```
instead of 
```
module test(
  output logic [31:0] b
);

  assign b = 32'hffffffff; // To hex for brevity

endmodule
```

AaronKel, 2019-05-24:
Also this line
`my_strtobin(data, endptr+2, -1, 2, case_type);`
should be
`my_strtobin(data, endptr+1, -1, 2, case_type);`

sumit0190, 2019-05-24:
Aah, that's where I went wrong - I took `endptr+2` instead of `endptr+1`.
And the `X` instead of the `z` was a typo - fixed that.

Now there's a different problem though. If you had, say, a 52-bit port, then `assign b = 52'bx` returns the following:
```
module \test
  attribute \src "testModule.v:2"
  wire width 52 output 1 \b
  connect \b 52'xxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxx
end
```
But with my change, `assign b='x` returns:
```
module \test
  attribute \src "testModule.v:2"
  wire width 52 output 1 \b
  connect \b 52'00000000000000000000xxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxx
end
```
This is probably because of line 127 in `const2ast.cc`, where we always blow up the size of the `data` vector to 32, although I'd have to play around a little more to be sure.

AaronKel, 2019-05-24:
"x" "X" "z" "Z" are all valid chars,
It looks like that case is to handle unsized constants, ideally it should get the value of the variable it is attached to.

sumit0190, 2019-05-25:
Hmm, the only way to do that, from what I could gather, is this:
The `assign` is handled at line 1451 of `genrtlil.cc`. Line 1454 eventually leads to `bitsAsConst` of `ast.cc`, where we zero pad the vector of bits depending on the size of the word. There we probably need some more logic to pad a vector like this, by checking the MSB of the argument that's passed in. Since the MSB will be `X` in the case I mentioned, it should pad it correctly.

I know I am missing some edge cases here though.

commit 34417ce5 (2019-05-27): Added support for unsized constants, fixes #1022
Includes work from @sumit0190 and @AaronKel
commit 40a070e2 (2019-05-27): Merge pull request #1043 from mmicko/unsized_constant

Added support for unsized constants, fixes #1022
closed 2019-05-27 by cliffordwolf