YosysHQ/yosys #1028: ABC assertion

Kmanfi, 2019-05-22:
## Steps to reproduce the issue
When running `abc -S 3 -lut 5` for some 3Mgates design... after some random time I'll get this.
Any idea how to debug this. If I run it from command line it works.
` /usr/local/bin/yosys-abc -s -f /tmp/yosys-abc-A0hxVw/abc.script`

```
2.4242.1. Executing ABC.
Running ABC command: <yosys-exe-dir>/yosys-abc -s -f <abc-temp-dir>/abc.script 2>&1
ABC: ABC command line: "source <abc-temp-dir>/abc.script".
ABC: 
ABC: + read_blif <abc-temp-dir>/input.blif 
ABC: + read_lut <abc-temp-dir>/lutdefs.txt 
ABC: + strash 
ABC: + ifraig 
ABC: + scorr 
ABC: Warning: The network is combinational (run "fraig" or "fraig_sweep").
ABC: + dc2 
ABC: + dretime 
ABC: + retime 
ABC: + strash 
ABC: + dch -f 
ABC: + if 
ABC: + mfs2 
ABC: + lutpack -S 3 
ABC: Assertion failed: (((unsigned)(ABC_PTRUINT_T)pFanin->pCopy) & 0xffff0000), function Lpk_CutTruth, file src/opt/lpk/lpkCut.c, line 200.
sh: line 1: 83118 Abort trap: 6           /usr/local/bin/yosys-abc -s -f /tmp/yosys-abc-A0hxVw/abc.script 2>&1
ERROR: ABC: execution of command "/usr/local/bin/yosys-abc -s -f /tmp/yosys-abc-A0hxVw/abc.script 2>&1" failed: return code 134.

```

## Expected behavior

No assertion.

## Actual behavior

Assertion.

Kmanfi, 2019-05-22:
Tried with `DISABLE_ABC_THREADS := 1`
and it works. I run it again to check was it false positive.

Kmanfi, 2019-05-22:
It was false positive, so still assertion.

cliffordwolf, 2019-05-23:
> If I run it from command line it works.
> `/usr/local/bin/yosys-abc -s -f /tmp/yosys-abc-A0hxVw/abc.script`

That's pretty weird. Are you sure it's not just a Heisenbug that comes and goes randomly?

> Any idea how to debug this.

Pretty much what you seem to have been trying anyways: Run the Yosys `abc` command with `-nocleanup -showtmp`, then continue with `yosys-abc` directly without Yosys in the way.

eddiehung, 2019-12-21:
Without providing a testcase there's not much any of us can do to help. At first glance it looks to be an upstream issue, since it fires an assertion inside abc. One workaround could be not use `lutpack` and see if it completes.

Please re-open if you have new information.

closed 2019-12-21 by eddiehung

gojimmypi, 2020-05-13:
@eddiehung I am seeing a similar error. (see also https://github.com/icebreaker-fpga/icebreaker-examples/issues/10)

```
2.44.1. Extracting gate netlist of module `\top' to `<abc-temp-dir>/input.blif'..
Extracted 61 gates and 85 wires to a netlist network with 24 inputs and 24 outputs.

2.44.1.1. Executing ABC.
Running ABC command: <yosys-exe-dir>/yosys-abc -s -f <abc-temp-dir>/abc.script 2>&1
ABC: <yosys-exe-dir>/yosys-abc: error while loading shared libraries: libreadline.so.6: cannot open shared object file: No such file or directory
ERROR: ABC: execution of command "/usr/local/bin/yosys-abc -s -f /tmp/yosys-abc-tlPlxn/abc.script 2>&1" failed: return code 127.
Makefile:79: recipe for target 'blinky.json' failed
make: *** [blinky.json] Error 1
```
when I run `yosys-abc -nocleanup -showtmp` I get an interesting message:
```
yosys-abc: error while loading shared libraries: libreadline.so.6: cannot open shared object file: No such file or directory
```
I have the dependencies installed per the [instructions](https://github.com/YosysHQ/yosys#setup), but perhaps something is still missing? Any idea how to find _what_ is missing?
 
As mentioned in https://github.com/YosysHQ/yosys/issues/643 I tried
```
sudo apt-get purge libreadline-dev
sudo apt-get install libreadline6-dev
```
and rebuilt yosys. I still see this error:
```
 $ yosys -ql 7seg_count.log  -p 'synth_ice40 -top top -json 7seg_count.json' 7seg_count.v
ERROR: ABC: execution of command "/usr/local/bin/yosys-abc -s -f /tmp/yosys-abc-HPMiJE/abc.script 2>&1" failed: return code 127.
```
Here's the tail end of the build, install, and test:
```
[100%] Building share/gowin/bram_init_16.vh
[100%] Building share/sf2/arith_map.v
[100%] Building share/sf2/cells_map.v
[100%] Building share/sf2/cells_sim.v

  Build successful.

gojimmypi@ElectronicsDesk : /mnt/c/workspace/yosys
0 $ sudo make install
[sudo] password for gojimmypi:
[Makefile.conf] CONFIG := gcc
mkdir -p /usr/local/bin
cp yosys yosys-config yosys-abc yosys-filterlib yosys-smtbmc /usr/local/bin
strip -S /usr/local/bin/yosys
strip /usr/local/bin/yosys-abc
strip /usr/local/bin/yosys-filterlib
mkdir -p /usr/local/share/yosys
cp -r share/. /usr/local/share/yosys/.
gojimmypi@ElectronicsDesk : /mnt/c/workspace/yosys
0 $ make test
[Makefile.conf] CONFIG := gcc
cd tests/simple && bash run-test.sh ""
make[1]: Entering directory '/mnt/c/workspace/yosys/tests/simple'
+ gcc -Wall -o /mnt/c/workspace/yosys/tests/tools/cmp_tbdata /mnt/c/workspace/yosys/tests/tools/cmp_tbdata.c
Test: arrays02 -> ok
Test: defvalue -> ERROR!
Note: Make sure that 'iverilog' is an up-to-date git checkout of Icarus Verilog.
../tools/autotest.mk:10: recipe for target 'defvalue.sv' failed
make[1]: *** [defvalue.sv] Error 1
make[1]: Leaving directory '/mnt/c/workspace/yosys/tests/simple'
Makefile:754: recipe for target 'test' failed
make: *** [test] Error 2
```
I have the latest `iverilog` installed, and `make check` seems to indicate it is happy:
```
 make check
make -C ivlpp all &&  make -C vhdlpp all &&  make -C vvp all &&  make -C vpi all &&  make -C libveriuser all &&  make -C cadpli all &&  make -C tgt-null all &&  make -C tgt-stub all &&  make -C tgt-vvp all &&  make -C tgt-vhdl all &&  make -C tgt-vlog95 all &&  make -C tgt-pcb all &&  make -C tgt-blif all &&  make -C tgt-sizer all &&  make -C driver all &&  true
make[1]: Entering directory '/mnt/c/workspace/iverilog/ivlpp'
make[1]: Nothing to be done for 'all'.
[... 71 lines cut ...]
make[1]: Leaving directory '/mnt/c/workspace/iverilog/tgt-vvp'
make[1]: Entering directory '/mnt/c/workspace/iverilog/tgt-vhdl'
make[1]: Nothing to be done for 'check'.
make[1]: Leaving directory '/mnt/c/workspace/iverilog/tgt-vhdl'
make[1]: Entering directory '/mnt/c/workspace/iverilog/tgt-vlog95'
make[1]: Nothing to be done for 'check'.
make[1]: Leaving directory '/mnt/c/workspace/iverilog/tgt-vlog95'
make[1]: Entering directory '/mnt/c/workspace/iverilog/tgt-pcb'
make[1]: Nothing to be done for 'check'.
make[1]: Leaving directory '/mnt/c/workspace/iverilog/tgt-pcb'
make[1]: Entering directory '/mnt/c/workspace/iverilog/tgt-blif'
make[1]: Nothing to be done for 'check'.
make[1]: Leaving directory '/mnt/c/workspace/iverilog/tgt-blif'
make[1]: Entering directory '/mnt/c/workspace/iverilog/tgt-sizer'
make[1]: Nothing to be done for 'check'.
make[1]: Leaving directory '/mnt/c/workspace/iverilog/tgt-sizer'
make[1]: Entering directory '/mnt/c/workspace/iverilog/driver'
make[1]: Nothing to be done for 'check'.
make[1]: Leaving directory '/mnt/c/workspace/iverilog/driver'
test -r check.conf || cp ./check.conf .
driver/iverilog -B. -BMvpi -BPivlpp -tcheck -ocheck.vvp ./examples/hello.vl
vvp/vvp -M- -M./vpi ./check.vvp | grep 'Hello, World'
Hello, World
```
Any suggestions will be appreciated.

daveshah1, 2020-05-13:
This is a totally unrelated problem so please continue any further discussion on a new issue. Please make sure you have tried a totally clean rebuild, and also include the output of `ldd /usr/local/bin/yosys-abc`.

nakengelhardt, 2020-05-13:
@gojimmypi I often see the "Note: Make sure that 'iverilog' is an up-to-date git checkout of Icarus Verilog." error, but it goes away when I run `make test` again. Haven't been able to figure out what causes it, but it's unrelated to abc problems. The second error you get after rebuild seems to be a different one (the line about libreadline is no longer there), have you tried running yosys-abc manually in the temp directory again to confirm if it's still the same error?

gojimmypi, 2020-05-13:
@daveshah1 and @nakengelhardt thanks for the suggestions. I've opened a new https://github.com/YosysHQ/yosys/issues/2048 for the problem I am seeing