YosysHQ/yosys #1033: Over-eager inference of distributed RAM for synth_xilinx

eddiehung, 2019-05-23:
## Steps to reproduce the issue

Unpacked arrays (even a 1-bit one) is inferred into distributed RAM.

`./yosys -p synth_xilinx dram.v -DUNPACKED`

```
module dram(input [3:0] i, output [4:0] o);
`ifdef UNPACKED
reg val [0:0];
`else
reg [0:0] val;
`endif
initial val[0] = 1'h1;
assign o = i + val[0];
endmodule
```

## Expected behavior

Result of: `./yosys -p synth_xilinx dram.v`
```
   Number of cells:                 10
     LUT1                            1
     MUXCY                           4
     XORCY                           5
```

## Actual behavior

```
   Number of cells:                 12
     LUT2                            2
     MUXCY                           4
     RAM64X1D                        1
     XORCY                           5
```

eddiehung, 2019-05-23:
Seems that `read_verilog` parses this as a `$memrd` cell right out of the gate... [show.pdf](https://github.com/YosysHQ/yosys/files/3213168/show.pdf)

daveshah1, 2019-05-23:
I don't think the problem is the `$memrd`, rather the thresholds in the DRAM mapping rules

eddiehung, 2019-05-23:
> I don't think the problem is the `$memrd`, rather the thresholds in the DRAM mapping rules

Correctamundo. It appears there is no `min bits` or `min efficiency` in the drams.txt rule book...

PR #1036 (2019-05-23): Add "min bits" and "min wports" to xilinx dram rules
Fixes #1033.

Logic behind `min wrports` is that there is no point inferring a hard ROM when soft LUTs are the ultimate ROM anyway, and leaving them as soft LUTs means it can be optimised by ABC.
closed 2019-05-23 by eddiehung