YosysHQ/yosys #1039: ABC takes a long time on larger circuit (LUT mapping)

cliffordwolf, 2019-05-25:
## Steps to reproduce the issue

Run `yosys -p synth_xilinx test.v` on the following design.

```
module test #(
        parameter integer ABITS = 12,
        parameter integer DBITS = 8
) (
        input CLK,
        input [ABITS-1:0] A1, A2, A3,
        input [DBITS-1:0] DI1, DI2, DI3,
        output [DBITS-1:0] DO1, DO2, DO3
);
        reg [DBITS-1:0] mem [0:(1<<ABITS)-1];
        always @(posedge CLK) begin
                mem[A1] <= DI1;
                mem[A2] <= DI2;
                mem[A3] <= DI3;
                DO1 <= mem[A1];
                DO2 <= mem[A2];
                DO3 <= mem[A3];
        end
endmodule
```

## Expected behavior

Produce a result in a reasonable time.

## Actual behavior

Spends a lot of time in ABC.

eddiehung, 2019-06-11:
Yosys is giving a pretty big circuit to ABC on which it is asking it do perform various optimisations (`dc2`) and compute structural choices (`dch`), both of which seem to be especially expensive:

```
2.21.1. Extracting gate netlist of module `\test' to `<abc-temp-dir>/input.blif'..
Extracted 184860 gates and 217688 wires to a netlist network with 32828 inputs and 36888 outputs.
```

eddiehung, 2019-06-12:
As a datapoint, vendor tools do not do much better:
```
synth_design: Time (s): cpu = 00:21:01 ; elapsed = 00:22:29 . Memory (MB): peak = 2429.168 ; gain = 1060.258 ; free physical = 323702 ; free virtual = 413611
```