YosysHQ/yosys #1053: SystemVerilog interface fields are implicitly defined?

pacak, 2019-05-29:
## Steps to reproduce the issue

```
cd tests/svinterfaces
yosys> read_verilog -sv svinterface1.sv
1. Executing Verilog-2005 frontend: svinterface1.sv
Parsing SystemVerilog input from `svinterface1.sv' to AST representation.
Generating RTLIL representation for module `\TopModule'.
svinterface1.sv:22: Warning: Identifier `\MyInterfaceInstance.mysig_out' is implicitly declared.
svinterface1.sv:25: Warning: Identifier `\MyInterfaceInstance.setting' is implicitly declared.
svinterface1.sv:27: Warning: Identifier `\MyInterfaceInstance.passThrough' is implicitly declared.
[skip]
```

This gets turned into an error with added `default_nettype none`.

## Expected behavior

No warning/error with `default_nettype none`

## Actual behavior

Warning/error.

Are there any reasons not to have default_nettype enabled in all the tests? As far as I understand setting it to none is a way to catch accidental typos and it's used all over the place...

Afromaniac, 2020-04-25:
I'd like to reaffirm that this is still an issue. 

I'm planning on putting a few hours into checking if this can be resolved without major structural changes. It seems to me like the introduced autowires are actaully correctly resolved later on in the process but seeing hundreds of lines worth of spurious implicit signal declaration warnings in verilator-approved code does not make me feel confident enough to use this language feature.

From what I can tell it was @rubund who implemented the SV Interface support in issue #597. Maybe he has some insights? I'm going to check if these warnings have always been part of SVI support or if they were introduced by some later commit.

Afromaniac, 2020-04-25:
Checked it out, this has been the behaviour since SVI were introduced which implies that it is nontrivial to simply 'fix'. I see two main ways of doing it:

**A**
In simplify.cc:1240:  Adding a special case to the `if (current_scope.count(str) == 0) {...}` block where local name resolution has failed.  Here we could check to see if the current identifier is some hierarchical name (`foo.bar`) - we could check for interface name resolution there. I don't fully grok the current namespace resolution code so maybe there's a better way of implementing this.

This feels a bit more like a hotfix than a true solution to be honest.

**B** (my favorite)
Check what interfaces a module contains before the simplification stage starts and explode them into wires immediately.

This makes the interface code into a preprocessing step so the later steps don't need to understand the interface abstraction. This does require us to be able to access the interface definition during simplification. ~~Are all modules parsed and added to the root AST_DESIGN node before individual modules are simplified?~~ Not all modules are parsed before simplification, nor are they added as children to a shared root AST_DESIGN node (which is the behaviour I expected).

If we went down this route I'd suggest (rough sketch) 

1. Make all modules/interfaces share a top level AST_DESIGN grandparent node.
2. When we encounter interfaces used in modules:  Check the AST_DESIGN grandparent for interface definitions.
3.If we find an interface child of AST_DESIGN with a name matching the one we encountered in 2) we explode the interface. Otherwise we can correctly error with "undefined interface".

If we want to be able to simplify modules one at a time the the way it is currently done this solution requires interface definitions to be read by `read_verilog` before they're used. I think it ends up a lot neater for a fairly low-impact criteria on the user. 

If we don't want to impose a new criteria on users we could change the code one step higher up and make `read_verilog` just parse source files into AST. Once the last read_verilog has been executed we call the `simplify` and `genrtlil` functions on the AST_DESIGN root. 

In the long run I like this option the most but it requires a bit of refactoring of code outside of the verilog preprocessor. I do think the AST to RTLIL logic should be more cleanly separated from the preprocessor to allow future preprocessor commands like `read_vhdl` to slot into the frontend->AST->RTLIL flow better.

I don't really know how feasible these suggestions are considered to be - could someone with a better macro perspective like @eddiehung or @cliffordwolf take a look? Getting proper reliable interface support (and later also multilang support) are fairly high priority for me. I'm not opposed to putting in the work of implementing this once we have a rough idea that'll actually work and be accepted.

rubund, 2020-04-25:
It has been a long time now, but I don't recall seeing those messages:
```
svinterface1.sv:22: Warning: Identifier `\MyInterfaceInstance.mysig_out' is implicitly declared.
...
```
when first implementing this.

The implementation of SV interfaces was done altering as little of existing code as possible - and up to the point where the test designs I had passed correctly.

It would be nice to see B implemented in the long run, but it sounds like a lot of work. Could "A" be done in the meantime?