YosysHQ/yosys #1056: Yosys output can not be disabled with pyosys

thasti, 2019-05-29:
## Steps to reproduce the issue

* Run the following python script:
```
from pyosys import libyosys as ys
```
The banner is shown (and there's no way to disable it)

* Run any command
```
from pyosys import libyosys as ys
d = ys.Design()
ys.run_pass("help", d)
```
The command output is shown (which is not always desirable)

* Run any command through the `tee -q` pass
```
from pyosys import libyosys as ys
d = ys.Design()
ys.run_pass("tee -q help", d)
```
The output of `help` is shown (which is not what happens in an interactive session)

## Comments
* While the C++ interface allows setting the output stream for logging, there's no way to do that using the python bindings (always uses stdout)
* Just having the banner is no big deal, but having `tee -q` to disable all the runtime output would be convenient, as long as there's no other python-exposed method for setting the log level

mmicko, 2019-05-29:
If you take a look at https://github.com/YosysHQ/yosys/blob/master/misc/py_wrap_generator.py#L2060 that is where stdout is set as output and few lines down banner is printed first time.
@btut maybe good idea would be to add methods to enable/disable output ?

cliffordwolf, 2019-05-30:
> maybe good idea would be to add methods to enable/disable output ?

I would even go as far as not printing the banner by default at all and provide a python wrapper for `Yosys::yosys_banner()`. @btut

pumbor, 2019-05-30:
Just came here to report this!

The `tee -q` fix is straightforward - in `TeePass`, `log_streams` needs to be backed up and restored in the same manner as `log_files`.

cliffordwolf, 2019-05-31:
> log_streams needs to be backed up and restored in the same manner as log_files

Thanks for pointing this out. Fixed in 90ec2cd.

thasti, 2019-05-31:
Thanks for looking at this, it's much appreciated. There already is a wrapper for `yosys_banner` available from Python, so it can probably go away from the initialization. 

```
>>> from pyosys import libyosys as ys
[automatic banner omitted]
>>> ys.yosys_banner()

 /----------------------------------------------------------------------------\
 |                                                                            |
 |  yosys -- Yosys Open SYnthesis Suite                                       |
 |                                                                            |
 |  Copyright (C) 2012 - 2018  Clifford Wolf <clifford@clifford.at>           |
 |                                                                            |
...
 |                                                                            |
 \----------------------------------------------------------------------------/

 Yosys 0.8+509 (git sha1 90ec2cda, gcc 8.3.0 -march=x86-64 -mtune=generic -O2 -fno-plt -fPIC -Os)

>>>
```

cliffordwolf, 2019-06-04:
ping @btut

closed 2019-06-05 by cliffordwolf

SalvatoreBarone, 2022-01-14:
Hi everyone. Please, are you so kind to tell me whether log can be turned off? Sometimes I do not want pyosys to print anything. Thank you in advance.

PR #1066 (2019-06-05): Remove yosys_banner() from python wrapper init
Fixes #1056
commit dd3c333c (2019-06-05): Remove yosys_banner() from python wrapper init, fixes #1056