YosysHQ/yosys #1058: Add support for attributes

mithro, 2019-05-31:
## Steps to reproduce the issue

The Verilog-2005 standard says;
> An attribute_instance can appear in the Verilog description as a prefix attached to a declaration, a module item, a statement, or a port connection. It can appear as a suffix to an operator or a Verilog function name in an expression.

This means for Yosys to be compatible with Verilog-2005 it should support attributes on the following (ones marked with `x` are already known to work);
- [x] Module declaration: `(* optimize *) module control ( ... );`
- [ ] Port declaration (input/output/inout): `(* port_attr *) input clk;`
- [x] Parameter: `(* param_attr *) parameter WIDTH = 10;`
- [x] Local parameter: `(* localparam_attr *) localparam MAGIC = 8'hFE;`
- [x] Net/Variable (wire/reg): `(* state_variable *) reg [7:0] S;`
- [ ] Port connection: `dff il (q, d, (* clock_line *) ckl, rst);`
- [ ] Suffix to an operator: `sum = a + (* ripple_adder *) b;`
- [ ] Suffix to a function call: `assign next_state = fsm_func(* meally_fsm *) (state, a, b, c);`
- [ ] Module instantiation: `(* inst_attr *) my_module my_instance (...);`

These examples are from [Verilog — 2001 - A Guide to the New Features of the Verilog® Hardware Description Language](https://link.springer.com/book/10.1007/978-1-4615-1713-9) Section [`25. Attributes`](https://link.springer.com/chapter/10.1007%2F978-1-4615-1713-9_27),

> ```verilog
> always @(state) //state machine with one-hot encoding
>   (* full_case, parallel_case *) case (state)
>   3'b001: next_state = 3'b010;
>   3'b010: next_state = 3'b100;
>   3'b100: next_state = 3'b001;
> endcase
> ```
>
> The attribute is specified as a prefix to a declaration, a module item, a statement or
> a port connection. Three examples are:
> ```verilog
> (* optimize *) module control ( ... );
>
> (* state_variable *) reg [7:0] S;
>
> dff il (q, d, (* clock_line *) ckl, rst};
> ```
>
> > The attribute is specified as a suffix to an operator or the function name in a function call.
> ```verilog
> sum = a + (* ripple_adder *) b;
>
> assign next_state = fsm_func(* meally_fsm *) (state, a, b, c);
> ```

These are the examples from the `3.8 Attributes` section of the [IEEE Standard for Verilog](http://staff.ustc.edu.cn/~songch/download/IEEE.1364-2005.pdf)

> #### Example 1—The following example shows how to attach attributes to a case statement:
> ```
> (* full_case, parallel_case *)
> case (foo)
> <rest_of_case_statement>
> (* full_case=1 *)
> (* parallel_case=1 *) // Multiple attribute instances also OK
> case (foo)
> <rest_of_case_statement>
> ```
>
> or
>
> ```verilog
> (* full_case, // no value assigned
> parallel_case=1 *)
> case (foo)
> <rest_of_case_statement>
> ```
>
> #### Example 2 — To attach the full_case attribute, but not the parallel_case attribute:
> ```verilog
> (* full_case *) // parallel_case not specified
> case (foo)
> <rest_of_case_statement>
> ```
>
> or
>
> ```verilog
> (* full_case=1, parallel_case = 0 *)
> case (foo)
> <rest_of_case_statement>
> ```
>
> #### Example 3—To attach an attribute to a module definition:
>
>
> ```verilog
> (* optimize_power *)
> module mod1 (<port_list>);
> ```
>
> or
>
> ```verilog
> (* optimize_power=1 *)
> module mod1 (<port_list>);
> ```
>
> #### Example 4—To attach an attribute to a module instantiation:
>
> ```verilog
> (* optimize_power=0 *)
> mod1 synth1 (<port_list>);
> ```
>
> #### Example 5—To attach an attribute to a reg declaration:
>
> ```verilog
> (* fsm_state *) reg [7:0] state1;
> (* fsm_state=1 *) reg [3:0] state2, state3;
> reg [3:0] reg1; // this reg does NOT have fsm_state set
> (* fsm_state=0 *) reg [3:0] reg2; // nor does this one
> ```
>
> #### Example 6—To attach an attribute to an operator:
>
> ```verilog
> a = b + (* mode = "cla" *) c;
> This sets the value for the attribute mode to be the string cla.
> ```
>
> #### Example 7—To attach an attribute to a Verilog function call:
>
> ```verilog
> a = add (* mode = "cla" *) (b, c);
> ```
>
> #### Example 8—To attach an attribute to a conditional operator:
>
> ```verilog
> a = b ? (* no_glitch *) c : d;
> ```

mithro, 2019-05-31:
There is more description in https://github.com/YosysHQ/yosys/issues/1009#issue-444016496 too.

mkurc-ant, 2019-05-31:
@mithro I'd like to work on making the Yosys parse those attributes. Regardless of whether their values are to be preserved throughout the synthesis flow.

daveshah1, 2019-05-31:
As far as I can tell (happy to accept counterexamples, although they should probably be issues of their own), the following also work:

 - Attributes on a port
```
module test (
	(* foo *) input clk
);
endmodule
```

- Suffix to an operator
```
module test (
	input a, input b
);
	wire x = a + (* foo *) b;
endmodule
```

- Suffix to a function call (*discarded but allowed*)
```
module test (
	output reg y
);
	function foo;
	begin
		foo = 1'b0;
	end
	endfunction

	always @* y = foo (* bar *) ();

endmodule
```

 - Module instantiation
```
module foo (
	input a
);
endmodule

module test (
	input a
);
	(* keep *) foo foo_i(.a(a));
endmodule
```

**Not** working:
 - Port connection
```
module foo (
	input a
);
endmodule

module test (
	input a
);
	 foo foo_i((* keep *) a);
endmodule
```

mkurc-ant, 2019-05-31:
I have a fix for the "attribute on port connection" case. https://github.com/YosysHQ/yosys/pull/1060

mithro, 2019-06-02:
@mkurc-ant Can you add these as tests?

mkurc-ant, 2019-06-03:
@mithro I added tests (not exactly those ones). See the PR #1060

mwkmwkmwk, 2020-02-01:
Given that #1060 is already merged, I think we can cross off the final TODO list item, and close this issue?

mwkmwkmwk, 2020-02-15:
Since noone disagrees, closing as per above.

closed 2020-02-15 by mwkmwkmwk