YosysHQ/yosys #1064: Fails to SAT solve, but why?

renau, 2019-06-04:
I have two verilog files. They should have the same result, but the SAT solver fails. Unclear to me why

## Steps to reproduce the issue

option1.v
```
module regfile2r1w_alt
  (input                   clk
  ,input [4-1:0]          waddr0
  ,input                  we0
  ,input [7-1:0]          din0
  ,input [4-1:0]          raddr0
  ,output [7-1:0]         q0
  ,output [7-1:0]         q1
  );

  reg [7-1:0] rf[2-1:0];

  reg [7-1:0] q0_next;
  always_comb begin
    if (we0 && raddr0 == waddr0)
      q0_next = din0;
    else
      q0_next = rf[raddr0];
  end

  always @(posedge clk) begin
    q0 <= q0_next;
  end

  always @(posedge clk) begin
    q1 <= q0_next;
  end

  always @(posedge clk) begin
    if (we0) begin
      rf[waddr0] <= din0;
    end
  end

endmodule
```

option2.v
```
module regfile2r1w_alt
  (input                   clk
  ,input [4-1:0]          waddr0
  ,input                  we0
  ,input [7-1:0]          din0
  ,input [4-1:0]          raddr0
  ,output [7-1:0]         q0
  ,output [7-1:0]         q1
  );

  reg [7-1:0] rf[2-1:0];

  reg [7-1:0] q0_next;
  always_comb begin
    if (we0 && raddr0 == waddr0)
      q0_next = din0;
    else
      q0_next = rf[raddr0];
  end
  always @(posedge clk) begin
    q0 <= q0_next;
  end

  reg [7-1:0] q1_next;
  always_comb begin
    if (we0 && raddr0 == waddr0)
      q1_next = din0;
    else
      q1_next = rf[raddr0];
  end
  always @(posedge clk) begin
    q1 <= q1_next;
  end

  always @(posedge clk) begin
    if (we0) begin
      rf[waddr0] <= din0;
    end
  end

endmodule
```

The script to check:
```
read_verilog -sv ./option1.v ; opt; prep; memory ; rename -top gold; design -stash gold;
read_verilog -sv ./option2.v ; opt; prep; memory ; rename -top gate; design -stash gate;
design -copy-from gold -as gold gold
design -copy-from gate -as gate gate

miter -equiv -make_assert -flatten gold gate equiv
hierarchy -top equiv
sat -prove-asserts -show-inputs -max 2 -verify -set-init-undef -ignore_unknown_cells equiv
```

ZipCPU, 2019-06-04:
Looking over your code, I found many fascinating details:

1. No initial value was given to either memory.  This could easily cause a mismatch between the two.

2. No initial values were given to either of the q0 or q1 registers.  This could also cause a mismatch.

3. Neither q0 nor q1 was declared as a register.  This generates a verific error, although none under non-verific

4. The addresses passed to the memory were too wide.  The memory should only see a single bit address.  This doesn't change a passing proof, but neither is it likely what the designer intended.

After fixing the above issues, I was able to prove that the two designs were equivalent using `abc pdr` from SymbiYosys, yet not with yosys' SAT solver.  (I seem to recall hearing something about SAT solvers not being able to handle memories property, although I can't remember the details.)  Indeed, I would recommend moving to SymbiYosys in general for any formal verification needs.

Dan

renau, 2019-06-04:
I understand that the code is not clean, but was created on purpose for testing.

The way that we found it is because doing this:

```
read_verilog -sv option1.v ; proc ; opt -fast ; pmuxtree ; memory_dff ; memory_share ; memory_collect
write_verilog option2.v
```

and then run the SAT solver it says that it is not equivalent.

Clearly something peculiar that yosys reads the original verilog, creates verilog but it is not able to check that they are equivalent.

cliffordwolf, 2019-06-05:
You should read the warning messages:

```
Setting up SAT problem:
Final constraint equation: { } = { }
Warning: Failed to import cell $techmap\gate.$memory\rf[0]$85 (type $dff) to SAT database.
Warning: Failed to import cell $techmap\gate.$memory\rf[1]$87 (type $dff) to SAT database.
Warning: Failed to import cell $techmap\gate.$procdff$77 (type $dff) to SAT database.
Warning: Failed to import cell $techmap\gold.$memory\rf[0]$30 (type $dff) to SAT database.
Warning: Failed to import cell $techmap\gold.$memory\rf[1]$32 (type $dff) to SAT database.
Warning: Failed to import cell $techmap\gold.$procdff$22 (type $dff) to SAT database.
```

Your `sat` command sets up a combinatorial sat problem, not a sequential one. Use for example `-seq N` to run a BMC with length N.

For example, the following passes fine:

```
read_verilog -sv ./option1.v ; opt; prep; memory ; rename -top gold; design -stash gold;
read_verilog -sv ./option2.v ; opt; prep; memory ; rename -top gate; design -stash gate;
design -copy-from gold -as gold gold
design -copy-from gate -as gate gate

miter -equiv -make_assert -make_outcmp -flatten gold gate equiv
hierarchy -top equiv
sat -prove-asserts -seq 10 -show-public -verify -set-init-undef -ignore_unknown_cells equiv
```

closed 2019-06-05 by cliffordwolf