YosysHQ/yosys #1065: opt_clean: Driver-driver conflict warning on unknown module

eddiehung, 2019-06-04:
## Steps to reproduce the issue

`./yosys -p "proc; opt -fast" conflict.v -q`

```
module top(input clk);
wire ce = 1'b1;
reg q = 1'b0;
always @(posedge clk)
    if (ce) q <= 1'b0;

(* keep *)
unknown_module u(.i(q));
endmodule
```

Produces the warning: `Warning: Driver-driver conflict for \q between cell u.i and constant 1'0 in top: Resolved using constant.`

## ~Expected~ Desired behavior

I've been thinking, and right now I can't see why this warning message would be helpful.
1. If connection was to an output port, then resolving it to be constant `1'b0` doesn't help the situation.
2. If it was an input port, then this is perfectly legitimate and nothing needed to be done anyway.

Is it reasonable to suppress this warning for unknown modules?

## Actual behavior

Produces the warning: `Warning: Driver-driver conflict for \q between cell u.i and constant 1'0 in top: Resolved using constant.`

#####

Actually, this came about when I was trying to create instantiations of temporary `$__name_` modules, and then when `opt` was run on them, I was getting these errors.

cliffordwolf, 2019-06-05:
Suppressing the warning is easy. See #1067. That that won't change the fact that we replace all instances of that net with 0.

> If connection was to an output port, then resolving it to be constant 1'b0 doesn't help the situation.

We don't replace the net with 1'b0 to "help the situation" with the conflict. It's just something that `opt_clean` inhenrently does when a net is driven by a constant. We can either do it silently or with a warning.

Note that the code printing the warning does not replace the net with the constant. This is extra code that analyses if opt_clean will replace a net with a constant and then prints a warning about it.

Note further that just suppressing to replace the constant on this one module port would help either. Instead we would need to suppress replacing the net with a constant on any port connected to that net, or other nets that alias them. That also means we would need to figure out which net-to-net connections to preserve. It's a deep rabbit hole.

eddiehung, 2019-06-05:
> We don't replace the net with 1'b0 to "help the situation" with the conflict.

I understand that, but the implication from the warning message was that the conflict was _resolved_:
`Warning: Driver-driver conflict for \q between cell u.i and constant 1'0 in top: Resolved using constant.`
However, I believe we can agree this is far from the truth.

closed 2019-06-05 by eddiehung

cliffordwolf, 2019-06-06:
> However, I believe we can agree this is far from the truth.

Yes. I'm open to suggestions if you have a better wording for the message than "resolved". How about:

```
Warning: Driver-driver conflict for \q between cell u.i and constant 1'0 in top: Using constant.
```

PR #1067 (2019-06-05): Suppress driver-driver conflict warning for unknown cell types
Fixes #1065
commit 8a6f9977 (2019-06-05): Suppress driver-driver conflict warning for unknown cell types, fixes #1065