YosysHQ/yosys #1069: Segfault when simulating

ZirconiumX, 2019-06-05:
## Steps to reproduce the issue

I haven't bisected this yet, but it is reproducible with 42190207 and a3a80b75.

[sim_segfault.zip](https://github.com/YosysHQ/yosys/files/3258802/sim_segfault.zip)

`yosys -p "read_verilog -sv sn74153_tb.sv; sim"` reproduces the segfault.

## Expected behavior

It should parse and run the simulation or fail gracefully. (Icarus thinks my code is invalid, but Verilator does not, which is...helpful).

## Actual behavior

```
Program received signal SIGSEGV, Segmentation fault.
std::vector<Yosys::hashlib::dict<Yosys::RTLIL::IdString, Yosys::RTLIL::Wire*, Yosys::hashlib::hash_ops<Yosys::RTLIL::IdString> >::entry_t, std::allocator<Yosys::hashlib::dict<Yosys::RTLIL::IdString, Yosys::RTLIL::Wire*, Yosys::hashlib::hash_ops<Yosys::RTLIL::IdString> >::entry_t> >::begin (this=0xa8)
    at /usr/bin/../lib/gcc/x86_64-linux-gnu/7.3.0/../../../../include/c++/7.3.0/bits/stl_vector.h:573
573           { return const_iterator(this->_M_impl._M_start); }
(gdb) bt full
#0  std::vector<Yosys::hashlib::dict<Yosys::RTLIL::IdString, Yosys::RTLIL::Wire*, Yosys::hashlib::hash_ops<Yosys::RTLIL::IdString> >::entry_t, std::allocator<Yosys::hashlib::dict<Yosys::RTLIL::IdString, Yosys::RTLIL::Wire*, Yosys::hashlib::hash_ops<Yosys::RTLIL::IdString> >::entry_t> >::begin (this=0xa8)
    at /usr/bin/../lib/gcc/x86_64-linux-gnu/7.3.0/../../../../include/c++/7.3.0/bits/stl_vector.h:573
No locals.
#1  0x0000000000a90cd9 in std::vector<Yosys::hashlib::dict<Yosys::RTLIL::IdString, Yosys::RTLIL::Wire*, Yosys::hashlib::hash_ops<Yosys::RTLIL::IdString> >::entry_t, std::allocator<Yosys::hashlib::dict<Yosys::RTLIL::IdString, Yosys::RTLIL::Wire*, Yosys::hashlib::hash_ops<Yosys::RTLIL::IdString> >::entry_t> >::empty (this=0xa8)
    at /usr/bin/../lib/gcc/x86_64-linux-gnu/7.3.0/../../../../include/c++/7.3.0/bits/stl_vector.h:760
No locals.
#2  0x0000000000ca6f99 in Yosys::hashlib::dict<Yosys::RTLIL::IdString, Yosys::RTLIL::Wire*, Yosys::hashlib::hash_ops<Yosys::RTLIL::IdString> >::empty (this=0x90) at ./kernel/hashlib.h:556
No locals.
#3  0x0000000000ca6f11 in Yosys::RTLIL::ObjIterator<Yosys::RTLIL::Wire*>::ObjIterator (this=0x7ffffffec0e0,
    list_p=0x90, refcount_p=0x88) at ./kernel/rtlil.h:421
No locals.
#4  0x0000000000ca4026 in Yosys::RTLIL::ObjRange<Yosys::RTLIL::Wire*>::begin (this=0x7ffffffec100)
    at ./kernel/rtlil.h:482
No locals.
#5  0x0000000000cd6b56 in (anonymous namespace)::SimInstance::SimInstance (this=0x181d4b0, shared=0x7ffffffec770,
    module=0x0, instance=0x0, parent=0x0) at passes/sat/sim.cc:96
        __range = @0x7ffffffec100: {list_p = 0x90, refcount_p = 0x88}
        __begin = {
          it = {<std::iterator<std::forward_iterator_tag, std::pair<Yosys::RTLIL::IdString, Yosys::RTLIL::Wire*>, long, std::pair<Yosys::RTLIL::IdString, Yosys::RTLIL::Wire*>*, std::pair<Yosys::RTLIL::IdString, Yosys::RTLIL::Wire*>&>> = {<No data fields>}, ptr = 0x7ffffffec108, index = 728}, list_p = 0x90, refcount_p = 0x88}
        __end = {
          it = {<std::iterator<std::forward_iterator_tag, std::pair<Yosys::RTLIL::IdString, Yosys::RTLIL::Wire*>, long, std::pair<Yosys::RTLIL::IdString, Yosys::RTLIL::Wire*>*, std::pair<Yosys::RTLIL::IdString, Yosys::RTLIL::Wire*>&>> = {<No data fields>}, ptr = 0x0, index = 124}, list_p = 0x5c0000006e, refcount_p = 0x7ffffde5c140 <main_arena+1280>}
#6  0x0000000000cd607f in (anonymous namespace)::SimWorker::run (this=0x7ffffffec770, topmod=0x0, numcycles=20)
    at passes/sat/sim.cc:691
No locals.
#7  0x0000000000cd5e1b in (anonymous namespace)::SimPass::execute (this=0x175e260 <(anonymous namespace)::SimPass>,
    args=warning: Type size unknown, assuming 1. Try casting to a known type, or void *.
warning: Type size unknown, assuming 1. Try casting to a known type, or void *.
Python Exception <class 'gdb.error'> Cannot perform pointer math on incomplete type "std::__cxx11::basic_string<char, std::char_traits<char>, std::allocator<char> >", try casting to a known type, or void *.:
std::vector of length 32, capacity 32, design=0x177d180) at passes/sat/sim.cc:858
        worker = {<(anonymous namespace)::SimShared> = {debug = false, hide_internal = true, writeback = false,
            zinit = false, rstlen = 1}, top = 0x0, vcdfile = <incomplete type>, clock = {
            hashtable = std::vector of length 0, capacity 0, entries = std::vector of length 0, capacity 0,
            ops = {<No data fields>}}, clockn = {hashtable = std::vector of length 0, capacity 0,
            entries = std::vector of length 0, capacity 0, ops = {<No data fields>}}, reset = {
            hashtable = std::vector of length 0, capacity 0, entries = std::vector of length 0, capacity 0,
            ops = {<No data fields>}}, resetn = {hashtable = std::vector of length 0, capacity 0,
            entries = std::vector of length 0, capacity 0, ops = {<No data fields>}}}
        numcycles = 20
        argidx = 1
        top_mod = 0x0
#8  0x00000000009f7b35 in Yosys::Pass::call (design=0x177d180, args=warning: Type size unknown, assuming 1. Try casting to a known type, or void *.
warning: Type size unknown, assuming 1. Try casting to a known type, or void *.
Python Exception <class 'gdb.error'> Cannot perform pointer math on incomplete type "std::__cxx11::basic_string<char, std::char_traits<char>, std::allocator<char> >", try casting to a known type, or void *.:
std::vector of length 32, capacity 32)
    at kernel/register.cc:235
        orig_sel_stack_pos = 1
        state = {parent_pass = 0x0, begin_ns = 140625000}
#9  0x00000000009f786d in Yosys::Pass::call (design=0x177d180, command=<incomplete type>) at kernel/register.cc:215
        args = warning: Type size unknown, assuming 1. Try casting to a known type, or void *.
warning: Type size unknown, assuming 1. Try casting to a known type, or void *.
Python Exception <class 'gdb.error'> Cannot perform pointer math on incomplete type "std::__cxx11::basic_string<char, std::char_traits<char>, std::allocator<char> >", try casting to a known type, or void *.:
std::vector of length 32, capacity 32
        cmd_buf = <incomplete type>
        tok = <incomplete type>
#10 0x0000000000ac106d in Yosys::shell (design=0x177d180) at kernel/yosys.cc:1173
        p = 0x17ae1c0 "sim"
        recursion_counter = 1
        command = 0x17ae1c0 "sim"
#11 0x00000000009c912f in main (argc=1, argv=0x7ffffffee0f8) at kernel/driver.cc:494
        frontend_command = <incomplete type>
        backend_command = <incomplete type>
        vlog_defines = warning: Type size unknown, assuming 1. Try casting to a known type, or void *.
warning: Type size unknown, assuming 1. Try casting to a known type, or void *.
std::vector of length 0, capacity 0
        passes_commands = warning: Type size unknown, assuming 1. Try casting to a known type, or void *.
warning: Type size unknown, assuming 1. Try casting to a known type, or void *.
std::vector of length 0, capacity 0
        plugin_filenames = warning: Type size unknown, assuming 1. Try casting to a known type, or void *.
warning: Type size unknown, assuming 1. Try casting to a known type, or void *.
std::vector of length 0, capacity 0
        output_filename = <incomplete type>
        scriptfile = <incomplete type>
        depsfile = <incomplete type>
        scriptfile_tcl = false
        got_output_filename = false
        print_banner = true
        print_stats = true
        call_abort = false
        timing_details = false
        mode_v = false
        mode_q = false
        opt = -1
```

eddiehung, 2019-06-05:
Looks like a null pointer dereference, caused by the lack of a top module. 
To fix it, try calling `hierarchy -auto-top` before `sim`.

#1072 makes Yosys more robust to this.

ZirconiumX, 2019-06-05:
Yep, that works; thank you!

PR #1072 (2019-06-06): Error out if no top module given before 'sim'
Fixes #1069
closed 2019-06-05 by ZirconiumX