YosysHQ/yosys #1080: python: SigBit data and offset attributes not available through python

pumbor, 2019-06-07:
The SigBit class is exposed through the python interface, and its members are `wire` and an anonymous union of `data` and `offset`. `py_wrap_generator.py` sees the enclosing {} around the union and treats it as its own namespace and therefore does not look for members within it, but the anonymous union members are part of the enclosing scope and can be treated like any other declared members for the purpose of that script. This works since the class members are accessed by set/get functions that reference the original C++ object.

I locally hacked py_wrap_generator.py to ignore the union enclosure and got the desired behavior with offset and data visible in SigBit objects, but it's not robustly written as I only have the one union in SigBit to test it against.

## Steps to reproduce the issue

```
cd yosys/examples/python-api
python3 -i script.py
cell = design.top_module().selected_cells()[0]
sigbit = cell.connections_[ys.IdString('\\A')].to_sigbit_vector()[0]
sigbit.offset
```

## Expected behavior

offset int value printed

## Actual behavior

```
Traceback (most recent call last):
  File "<stdin>", line 1, in <module>
AttributeError: 'SigBit' object has no attribute 'offset'
```

## py_wrap_generator.py hack
		# for anonymous unions, ignore union enclosure by skipping start line and replacing end line with new line
		if 'union {' in line:
			j = i+1
			while j < len(source_text):
				union_line = source_text[j]
				if '};' in union_line:
					source_text[j] = '\n'
					break
				j += 1
			if j != len(source_text):
				i += 1
				continue

cliffordwolf, 2019-06-07:
cc @btut

btut, 2019-07-03:
Hi,
sorry for the late reply. Your hack seems to be appropriate, could you create a PR?

commit 177a7cb2 (2020-01-29): Merge pull request #1510 from pumbor/master

handle anonymous unions to fix #1080
commit 1e92e2d1 (2020-01-29): handle anonymous unions to fix #1080
closed 2020-01-29 by nakengelhardt