YosysHQ/yosys #1083: Disable test

Kmanfi, 2019-06-10:
## Steps to reproduce the issue
Python [http://www.myhdl.org](url) produce code like the example below.
```
module disa_test (a, b, oo);
    input a, b;
    output oo;
    reg c;

    always @(a) begin : aa
    c = a;
    disable aa;
    c = b;
    end
assign oo = c;
endmodule

```

## Expected behavior

Should read.
## Actual behavior

 ERROR: syntax error, unexpected TOK_ID

cliffordwolf, 2019-06-20:
> Should read.

I wonder what kind of circuit you'd expect this to synthesize into.

eddiehung, 2019-06-27:
@Kmanfi Can you provide a little more context as to when/why myhdl outputs the disable statement? Is it intended to be synthesisable?

DaveMcEwan, 2023-10-16:
Verilog doesn't have the `break` or `continue` keywords so the `disable` keyword is supposed to fulfill both of those ideas. The keywords `break` and `continue` were introduced to SystemVerilog in IEEE 1800-2005. Repeating examples from IEEE 1364 here for posterity, with slight modifications for readability. In IEEE 1364-2001, these appear in clause 11, pages 162 and 163. In IEEE 1364-2005, these appear in clause 10.3, pages 150 and 151:

Example 1 This example illustrates how a block disables itself.
```verilog
begin: l_foo
  rega = regb;
  disable l_foo;
  regc = rega; // This assignment will never execute.
end
```

Example 2 This example shows the disable statement being used within a named block in a manner similar to a forward _goto_. The next statement executed after the disable statement is the one following the named block.
```verilog
begin: l_foo
  ... statements ...
  if (a == 0)
    disable l_foo;
  ... statements ...
end // End of named block (l_foo).
// Continue with code following named block.
```

Example 3 - Early return from a task isn't generally synthesisable.

Example 4 This example shows the disable statement being used in an equivalent way to the two statements `continue` and `break` in the C programming language. The example illustrates control code that would allow a named block to execute until a loop counter reaches `n` iterations or until the variable `a` is set to the value of `b`. The named block `l_break` contains the code that executes until `a == b`, at which point the `disable l_break;` statement terminates execution of that block. The named block `l_continue` contains the code that executes for each iteration of the loop. Each time this code executes the `disable l_continue;` statement, the block terminates and execution passes to the next iteration of the for loop.
```verilog
begin: l_break
  for (i = 0; i < n; i = i+1) begin: l_continue
    
    if (a == 0) // In C terminology, "continue" the loop
      disable l_continue;
    ... statements ...
  
    if (a == b) // In C terminology, "break" from the loop
      disable l_break;
    ... statements ...

  end // End of l_continue
end // End of l_break
```
Examples 5 and 6 are not generally synthesisable.

The example given by @Kmanfi is equivalent to `always @(a) c = a;`, i.e. everything in the process after the `disable` is irrelevant but must be able to compile/elaborate.

Could Yosys support the `disable` keyword in those situations, and the semantically equivalent to `break` and `continue` keywords in SystemVerilog mode? One approach could be to (internally) rewrite the loop with additional variables and conditions giving the same behaviour.