YosysHQ/yosys #1087: Memory inference fails for equivalent behaviour for read-after-write

eddiehung, 2019-06-11:
## Steps to reproduce the issue

` ./yosys -p "synth -run coarse; memory_bram -rules +/xilinx/brams.txt; stat" ram1.v -f "verilog -DBUG"`

```
module reg_file(input wire clk,
                input wire wr_en,
                input wire [9:0] w_addr,
                input wire [9:0] r_addr,
                input wire [17:0] w_data,
                output reg [17:0] r_data,
                output reg [17:0] r_data2);

  reg [17:0] ram [0:1023];

  always @(posedge clk)
  begin
    r_data <= ram[r_addr];
  end

  always @(posedge clk)
  begin
`ifdef BUG
    r_data2 <= ram[w_addr];
    if (wr_en) begin
      ram[w_addr] <= w_data;
      r_data2 <= w_data;
    end
`else
    if (wr_en) begin
      ram[w_addr] <= w_data;
      r_data2 <= w_data;
    end
    r_data2 <= ram[w_addr];
`endif
  end

endmodule
```

## Expected behavior

Successful inference (as happens when `-DBUG` is omitted)
```
   Number of cells:                 31
     $__XILINX_RAMB18_SDP            4
     $dff                            4
     $eq                             8
     $mux                           13
     $pmux                           2
```

## Actual behavior

Failed inference
```
   Number of cells:                  4
     $dff                            1
     $mem                            1
     $mux                            2
```

eddiehung, 2019-06-11:
Testcase derived from dual port RAM template: https://github.com/alexforencich/verilog-cam/blob/master/rtl/ram_dp.v

daveshah1, 2019-06-19:
Yosys currently expects a register on the address port to infer read-after-write memory:
```
module reg_file(input wire clk,
                input wire wr_en,
                input wire [9:0] w_addr,
                input wire [9:0] r_addr,
                input wire [17:0] w_data,
                output [17:0] r_data);

  reg [17:0] ram [0:1023];
  reg [9:0] r_addr_reg;
  always @(posedge clk)
  begin
    r_addr_reg <= r_addr;
    if (wr_en) begin
      ram[w_addr] <= w_data;
    end
  end

  assign r_data = ram[r_addr_reg];
endmodule
```

For comparison, your example produces:
[image]

`memory_dff` would need to be modified to recognise this mux and make a transparent port. But I think there could be a semantics issue here as we move towards supporting TDP RAM - unlike the current semantics where transparent read ports are considered read-after-write for all write ports, this kind of Verilog specifies read-after-write for same-port accesses only. Not sure whether cross-port accesses might be into UB anyway though.

PR #3189 (2022-05-18): Add memory_libmap pass.
Fixes #1959.
Fixes #1134.
Fixes #1958.
Fixes #1836.
Fixes #1101.
Fixes #1087.
Fixes #1088.
Fixes #1802.
Fixes #1550.
Fixes #3205.
closed 2022-05-18 by mwkmwkmwk