YosysHQ/yosys #1089: Getting Gates and Transistor count

mickey4u, 2019-06-11:
I am new to Yosys and synthesis but what I want to achieve is to get the number of gates and transistors after the design unit has undergone systhesis. How do I achieve that?

ZipCPU, 2019-06-11:
A couple of steps:

1. Read your files into yosys, using `read -sv filename.sv`.  VHDL files may be read similarly when using the SymbioticEDA Suite.

2. Synthesize your design.  For generic architectures, such as general ASIC gate counts, `synth -top topmodulename` is usually sufficient.

3. You'll then want to map to gates, using `abc -g GATES`.  If you want some kind of special mapping, such as mapping to NAND and NOT gates plus FFs, then you can use `abc -g CMOS2`.  See the help for `abc` within yosys for more information.

4. Finally, run the `stat` command to get the numbers of the various items used.

That should get you a number of gates.  You'll need to do the gate to transistor mapping yourself.

Feel free to check out [this article on ZipCPU.com for an example](http://zipcpu.com/zipcpu/2019/03/28/return-decoding.html).

Dan

mickey4u, 2019-06-11:
@ZipCPU Thanks. So Gates to transistor mapping is not supported. Any direction on how to achieve that.

ZipCPU, 2019-06-11:
Depends ... how are the gates being implemented by your foundry?  Seems like you'd kind of need to know that before you could go all the way to transistors.

mickey4u, 2019-06-11:
@ZipCPU Okay. I will use the number of gates for now. What I want to achieve is, if I have multiple implementation of say an adder how do I select one the uses few number of gates or transistors.

eddiehung, 2019-06-11:
Yosys maps to gates (via ABC) described in the Liberty format. The Liberty format has a attribute "area" which I think is used by ABC to make area-delay mapping decisions -- I presume this is the attribute used to capture transistor counts, so I suggest looking down that road? 
You could look at the Yosys gate counts and multiply them by this area value, and/or you can get similar information from ABC, using ABC commands like `print_stats`...

mickey4u, 2019-06-11:
@eddiehung Will do. Thanks

cliffordwolf, 2019-06-20:
The Yosys `stat` command can also read a liberty file. If called with a liberty file it will output area numbers for the design.

closed 2019-06-20 by cliffordwolf