YosysHQ/yosys #1090: python: wire() and cell() methods of Module segfault when wire/cell doesn't exist

pumbor, 2019-06-11:
Module::wire() and Module::cell() look up a wire or cell by IdString, and return nullptr if not found. The associated python wrappers don't account for the nullptr case and immediately dereference the returned pointer, resulting in a segfault.

## Steps to reproduce the issue
```
cd yosys/examples/python-api
python3 -i script.py
module.wire(ys.IdString('\\not_a_wire'))
```

## Expected behavior

Not a segfault. Aside from that, I'm not really sure. From a python perspective the ideal behavior would be to return `None`, but I don't see a way to shoehorn that in with boost::python. Maybe the wrapper classes need to be updated to tolerate a null ref_obj, with a function exposed to python to query whether it is null.

If the methods are removed from the python interface, then we need some other way to cheaply check for the existence of a wire. Module.wires_ is accessible through python, but requires copying the entire dictionary.

## Actual behavior
```
Segmentation fault (core dumped)
```

daveshah1, 2019-06-11:
If this helps, this is how I approached a similar problem with the nextpnr Python bindings:

https://github.com/YosysHQ/nextpnr/blob/master/common/pywrappers.h#L132-L137

https://github.com/YosysHQ/nextpnr/blob/master/common/pywrappers.h#L158-L167

The idea is to check for a nullptr before de-referencing, and throw an exception if null. This is then caught by the final layer of wrapping which returns an empty `boost::python::object` (this becomes a Python `None`) instead of the derefenced entity.

cliffordwolf, 2019-06-21:
cc @btut

btut, 2019-07-04:
Sorry for the long delay. This should fix your issue.
Thanks for the suggestion @daveshah1. Unfortunately, the conversion to the actual boost::python::object is done by the library itself. I just convert to the wrapped yosys classes. Therefore, I cannot return a None object.
Nevertheless I now introduced that same check and a runtime exception will be thrown, however this exception is not caught in my code (as you did for nextpnr) but will be converted to a python exception and forwarded for the user to handle.

PR #1159 (2019-07-05): Throw runtime exception when trying to convert inexistend C++ object to Python
Throw runtime exception when trying to convert a c++-pointer to a python-object in case the pointer is a nullptr to avoid a segfault.

Fixes #1090
commit 3a1a41bd (2019-07-05): Throw runtime exception when trying to convert a c++-pointer to a
python-object in case the pointer is a nullptr to avoid a segfault.

Fixes #1090
closed 2019-07-05 by cliffordwolf