YosysHQ/yosys #1096: Yosys only checks the local directory for $readmemh files

ZipCPU, 2019-06-14:
## Steps to reproduce the issue

Create a project with a subdirectory, `subd`.  Place the following mcve into that subdirectory:

```verilog
module	memdev(i_clk, i_addr, o_data);
	parameter	DW=32;
	parameter	HEXFILE="rom.hex";
	input	wire			i_clk;
	input	wire	[4:0]  	i_addr;
	output	reg	[(DW-1):0]	o_data;

	reg	[(DW-1):0]	mem	[0:31];
	initial	$readmemh(HEXFILE, mem);
	always @(posedge i_clk)
		o_data <= mem[sw_addr];

endmodule
```

Also, place the `rom.hex` file into the same subdirectory.

Now imagine that this Verilog file is part of a larger project with many pieces.  You might wish to synthesize it from the basic directory rather than the subdirectory using a command such as,

`yosys -p 'read -sv subd/mcve.v; synth'`

## Expected behavior

I would expect an option to have the hex file read from either the local directory, the directory that the `.v` file is found within, or otherwise a configurable parameter indicating where the associated data files might be found.

## Actual behavior

Yosys currently checks files relative to the location where yosys was called only.  As a result, it cannot find this hex file.

AaronKel, 2019-06-16:
Sounds like $readmemh is following 22.4 of 1800-2017

> The filename is the name of the file to be included in the source file. The filename can be a full or relative path name.
>
> The filename can be enclosed in either quotes or angle brackets, which affects how a tool searches for the file, as follows: 
> 
> - When the filename is enclosed in double quotes ("filename"), for a relative path the compiler’s current working directory, and optionally user-specified locations are searched. 

Maybe try adding (If not already)

> - Idir
> 
>         add 'dir' to the directories which are used when searching include
>         files

AaronKel, 2019-06-18:
Sorry I had some wrong information, it seems like -Idir is only in the read_verilog case. 
`yosys -p 'read -incdir subd -sv subd/mcve.v; synth'`
Works for me

smunaut, 2021-05-27:
Just adding +1 that this would be useful.  `-I` (for `read_verilog` and `-incdir` for `read` don't work. Those only work for include files and not for files used by $readmemh).

lethalbit, 2022-08-22:
As of writing, commit c26b2bf5 `$readmem` has [logic](https://github.com/YosysHQ/yosys/blob/c26b2bf543a226e65a3fb07040bb278d668accf2/frontends/ast/simplify.cc#L4306) for trying first in the tool CWD followed by checking local to where the file is.

Tentatively closing as resolved as with a test case similar to the original post's example it works as expected.

closed 2022-08-22 by lethalbit