YosysHQ/yosys #1103: Memory leak in verilog frontend (attribute related)

eddiehung, 2019-06-18:
## Steps to reproduce the issue

On 64947453: `valgrind --leak-check=full ./yosys picorv32.v`

## Expected behavior

No leak.

## Actual behavior

Pointing to this line: https://github.com/YosysHQ/yosys/blob/64947453e22404e570d670416ad66511a799e666/frontends/verilog/verilog_parser.y#L217
```
==13571== 1,008 bytes in 21 blocks are definitely lost in loss record 2 of 2
==13571==    at 0x4C3017F: operator new(unsigned long) (in /usr/lib/valgrind/vgpreload_memcheck-amd64-linux.so)
==13571==    by 0x5E6581: frontend_verilog_yyparse() (verilog_parser.y:217)
==13571==    by 0x5F349C: Yosys::VerilogFrontend::execute(std::istream*&, std::__cxx11::basic_string<char, std::char_traits<char>, std::allocator<char> >, std::vector<std::__cxx11::basic_string<char, std::char_traits<char>, std::allocator<char> >, std::allocator<std::__cxx11::basic_string<char, std::char_traits<char>, std::allocator<char> > > >, Yosys::RTLIL::Design*) (verilog_frontend.cc:447)
==13571==    by 0x52E6CF: Yosys::Frontend::execute(std::vector<std::__cxx11::basic_string<char, std::char_traits<char>, std::allocator<char> >, std::allocator<std::__cxx11::basic_string<char, std::char_traits<char>, std::allocator<char> > > >, Yosys::RTLIL::Design*) (register.cc:373)
==13571==    by 0x52F380: Yosys::Frontend::frontend_call(Yosys::RTLIL::Design*, std::istream*, std::__cxx11::basic_string<char, std::char_traits<char>, std::allocator<char> >, std::vector<std::__cxx11::basic_string<char, std::char_traits<char>, std::allocator<char> >, std::allocator<std::__cxx11::basic_string<char, std::char_traits<char>, std::allocator<char> > > >) (register.cc:493)
==13571==    by 0x52F187: Yosys::Frontend::frontend_call(Yosys::RTLIL::Design*, std::istream*, std::__cxx11::basic_string<char, std::char_traits<char>, std::allocator<char> >, std::__cxx11::basic_string<char, std::char_traits<char>, std::allocator<char> >) (register.cc:471)
==13571==    by 0x5873D6: Yosys::run_frontend(std::__cxx11::basic_string<char, std::char_traits<char>, std::allocator<char> >, std::__cxx11::basic_string<char, std::char_traits<char>, std::allocator<char> >, std::__cxx11::basic_string<char, std::char_traits<char>, std::allocator<char> >*, std::__cxx11::basic_string<char, std::char_traits<char>, std::allocator<char> >*, Yosys::RTLIL::Design*) (yosys.cc:995)
==13571==    by 0x51D88D: main (driver.cc:505)
```

cliffordwolf, 2019-06-21:
Looks like ec4565009ae69409eb01f1b595f5f59fcc969ce2 fixed this. Not entirely sure why tho. I'll have a closer look..

cliffordwolf, 2019-06-21:
The memory leak was added in ce4a0954b and fixed in ec45650.

It effected only non-Verilog-2000-style parameters. I.e. no memory leak:

```
module test #( parameter FOO = 42 );
endmodule
```

But memory leak:

```
module test;
  parameter FOO = 42;
endmodule
```

We use Verilog-2000-style parameters in most tests, thus most of our tests would not have trigger it.

closed 2019-06-21 by cliffordwolf