YosysHQ/yosys #1126: Wrong bit vector indexing with constant function

cassiersg, 2019-06-23:
## Steps to reproduce the issue

When Yosys is given verilog code that indexes a bit vector with a constant function, its output generation is wrong.
For instance:
```
module test(c);

input [9:0] c;

function integer a(input integer b);
    begin
        a = b+1;
    end
endfunction

localparam f = 5;
wire [a(f)-1:0] d;
assign d = c[a(f)-1:0]; // wrong behavior
localparam e = a(f);
// assign d = c[e-1:0]; // correct behavior
```
The yosys (git fb8fab4a2) invocation:
```
./yosys -p "read_verilog test.v; write_verilog test_synth.v; write_json test_synth.json"
```

*Provide instructions for reproducing the issue. Make sure to include
all necessary source files. (You can simply drag&drop a .zip file into
the issue editor.)*

## Expected behavior

Verilog output `test_synth.v` should be similar to the result obtained when the "wrong" line is replaced by the "correct" one (i.e., the function computation is moved to a localparam):
```
module test(c);
  (* src = "test.v:3" *)
  input [9:0] c;
  (* src = "test.v:12" *)
  wire [5:0] d;
  assign d = c[5:0];
endmodule
```

## Actual behavior

 `test_synth.v` contains (json output exhibits the same issue):
```
[...]
wire [9:0] _10_ = c;                                                                                                                                                                                         
assign _7_ = _10_[$signed(32'd0) +: 2];  
assign d = { 4'h0, _7_ };  
[...]
```

## Attached files

Synthesis output files with the "wrong" and the "correct" indexing.

[test.zip](https://github.com/YosysHQ/yosys/files/3318039/test.zip)

cliffordwolf, 2019-06-24:
Had a quick look at this issue. Here are some notes:
- The problem is that in `foo[A:B]` the expressions `A` and `B` must be constant. (Unlike `foo[A+:B]` where only B must be constant.)
- There are two mechanisms for evaluating functions in the `read_verilog` front-end. One for constant functions and one for non-constant functions.
- For some reason the front-end expands this using the mechanism for non-constant functions.
- Going through `localparam e = a(f);` forces the use of the mechanism for constant expressions.

zachjs, 2021-03-01:
This was fixed as part of #2573. I now get the expected version with:
```Verilog
wire [5:0] d;
assign d = c[5:0];
```

closed 2021-03-01 by zachjs