YosysHQ/yosys #1132: wreduce causes muxcover to not identify covering opportunity

eddiehung, 2019-06-25:
## Steps to reproduce the issue

` ./yosys -p "proc; opt; wreduce; simplemap; muxcover; dump" mux_if_bal_3_1.v`

```
module mux_if_bal_3_1 #(parameter N=3, parameter W=1) (input [N*W-1:0] i, input [$clog2(N)-1:0] s, output reg [W-1:0] o);
always @* begin
    o <= {{W{{1'bx}}}};
    if (s[0] == 1'b0)
     if (s[1] == 1'b0)
      o <= i[0*W+:W];
     else
      o <= i[1*W+:W];
    else
     if (s[1] == 1'b0)
      o <= i[2*W+:W];
end
endmodule
```

## Expected behavior

Pack into `$_MUX4_`

## Actual behavior

No packeroonies.

Before `wreduce`:
[image]

After `wreduce`:
[image]

eddiehung, 2019-06-25:
@cliffordwolf As discussed, you mentioned that `wreduce` had some special handling for mux cells that we could make optional, so that these opportunities are not lost...

cliffordwolf, 2019-06-26:
Thanks for this example. I think there's no need to do anything to wreduce here. It's just that the default costs in muxcover are set to values that don't incentivize muxcover to use a MUX4 to replace two MUX. From `help muxcover`:

```
    -mux4[=cost], -mux8[=cost], -mux16[=cost]
        Use the specified types of MUXes (with optional integer costs). If none
        of these options are given, the effect is the same as if all of them are.
        Default costs: $_MUX_ = 100, $_MUX4_ = 220,
                       $_MUX8_ = 460, $_MUX16_ = 940
```

And after wreduce has eliminated one of the muxes, muxcover has the choice to replace two MUX2 with one MUX4, but doesn't do it because two MUX2 have lower cost. (Before wreduce muxcover thinks it replaces three MUX2 with one MUX4, which it happily does.)

Setting the MUX4 cost to under 200 results in the desired output:

```
yosys -p "proc; opt; wreduce; simplemap; muxcover -mux4=150;; show" mux_if_bal_3_1.v
```

[image]

However, you can also call `wreduce` with `-keepdc` to disable this particular optimization:

```
./yosys -p "proc; opt; wreduce -keepdc; simplemap; muxcover;; show" mux_if_bal_3_1.v
```

But his would not help in the general case where two MUX2 should be replaced with one MUX4:

```
module test (input S, T, input A, B, C, output Y);
  assign Y = S ? A : T ? B : C;
endmodule
```

Will result in two MUX2 with `yosys -p "proc; opt; simplemap; muxcover;; show" test.v` and one MUX4 with `yosys -p "proc; opt; simplemap; muxcover -mux4=150;; show" test.v`.

The current muxcover costs predate the muxcover ability to use partial matches and arguably should be lowered. Please feel free to lower the default costs to better suite your use-case.

eddiehung, 2019-06-26:
@cliffordwolf Thanks for the explanation, that makes perfect sense. 

The problem with blowing a mux away is that a 5:1 mux now costs the same as a 4:1 mux, and if my threshold of inferring muxes is >4, then I would have missed this opportunity. But one could argue that opportunity never existed anyway, so let me think about this some more. Perhaps I need to change my testsuite to consider this a little more...

I'm wondering if `-keepdc` has a big impact on other cells, such that if I were to use this flag when I want to do mux inference, would I miss some obvious opportunities?

eddiehung, 2019-06-26:
Okay looks like I might have oversimplified the testcase. Now let's consider a 5:1 multiplexer:

```
module mux_if_bal_5_1 #(parameter N=5, parameter W=1) (input [N*W-1:0] i, input [$clog2(N)-1:0] s, output reg [W-1:0] o);
always @* begin
    o <= {{W{{1'bx}}}};
    if (s[0] == 1'b0)
     if (s[1] == 1'b0)
      if (s[2] == 1'b0)
       o <= i[0*W+:W];
      else
       o <= i[1*W+:W];
     else
      if (s[2] == 1'b0)
       o <= i[2*W+:W];
      else
       o <= i[3*W+:W];
    else
     if (s[1] == 1'b0)
      if (s[2] == 1'b0)
       o <= i[4*W+:W];
end
endmodule
```

which I can throw at:
`yosys -p "proc; opt; wreduce; debug muxcover -mux8=350"`

to get:
```
7. Executing MUXCOVER pass (mapping to wider MUXes).
Covering MUX trees in module mux_if_bal_5_1..
  Treeifying 4 MUXes:
    Found tree with 4 MUXes at root \o.
    Finished treeification: Found 1 trees.
  Covering trees:
    Populating cache of decoder muxes.
        Decode cost for mux2 at $procmux$18_Y: 0
        Best cost for \i [1]: 0
        Best cost for \i [0]: 0
      Cost of mux2 at $procmux$18_Y: 100
        Decode cost for mux8 at $procmux$18_Y: 90
        Best cost for \i [1]: 0
        Best cost for \i [0]: 0
      Cost of mux8 at $procmux$18_Y: 440
<<SNIP>>
```
Surely using a MUX8 to implement a MUX2 shouldn't require any decoding?

cliffordwolf, 2019-06-27:
> Surely using a MUX8 to implement a MUX2 shouldn't require any decoding?

Yup, that's a bug. Fix pending in #1142 (https://github.com/YosysHQ/yosys/pull/1142/commits/69d810e4a831fc1f17d886d7f371d9785e9022e7#diff-cd9eed159fba9a54c4b10e58df381084R164).

commit 69d810e4 (2019-06-27): Fix handling of partial covers in muxcover, fixes #1132
PR #1142 (2019-06-27): Fix handling of partial covers in muxcover
Fixes #1132
closed 2019-06-27 by eddiehung