YosysHQ/yosys #1134: Synthesis of asymmetric block RAMs

whitequark, 2019-06-25:
Dual-port block RAMs with asymmetric read and write ports are very useful CDC primitives. Or... would be if one didn't have to instantiate them, losing the ability to use a generic async FIFO implementation.

It is my understanding that neither the `memory_collect` nor the `memory_bram` pass currently support asymmetric block RAMs at all. What would be the steps to fix that? I believe the `$mem` cell would need to be extended first, or possibly there would be a separate `$asymmem` cell.

Note that I am **not** talking about asymmetric block RAM inference from Verilog. I am perfectly happy to emit `$memrd` and `$memwr` cells with asymmetric widths directly from nMigen. (Is it even possible to describe an asymmetric RAM in Verilog? Does any vendor try to infer them at all?)

cliffordwolf, 2019-06-26:
> Note that I am not talking about asymmetric block RAM inference from Verilog.

Nevertheless I think the best way to approach this is to come up with a Verilog pattern we want to be able to infer asymmetric block RAMs from. nMigen can then simply target the equivalent RTLIL pattern.

> Is it even possible to describe an asymmetric RAM in Verilog? Does any vendor try to infer them at all?

Afaik there's no Verilog feature specifically for describing asymmetric RAMs. I don't know what Vendor tools do. Researching this would probably be a good first step.

whitequark, 2019-06-26:
> Nevertheless I think the best way to approach this is to come up with a Verilog pattern we want to be able to infer asymmetric block RAMs from. nMigen can then simply target the equivalent RTLIL pattern.

But that's not "simply": the simple thing to do is to emit `$memrd` and `$memwr` cells with appropriate address and data widths, isn't it?

cliffordwolf, 2019-06-27:
> But that's not "simply": the simple thing to do is to emit $memrd and $memwr cells with appropriate address and data widths, isn't it?

Well, that's possibly the most simple thing to do on the emit side, but not necessarily the most simple solution holistically because it would be a big change to everything downstream that processes those cells.

We've been discussing single-port RAMs recently and the path forward seems to be to either extend memory_bram, or write a replacement, but not change the `$mem*` cells.

Similarly, asymmetric block RAMs could be modeled using a scheme such as the following (untested code):

```
input we;
input [8:0] waddr;
input [7:0] wdata;

input [7:0] raddr;
output reg [15:0] rdata;

reg [15:0] mem [0:255];

always @(posedge clock) begin
  if (we) begin
    case (waddr[0])
      1'b0: mem[waddr[8:1]][ 7:0] <= wdata;
      1'b1: mem[waddr[8:1]][15:8] <= wdata;
    endcase
  end
  rdata <= mem[raddr];
end
```

And the bram mapper could recognize this and map it directly to an asymmetric bram primitive, without having to change each and every consumer of `$mem*` to know about asymmetric brams.

whitequark, 2019-06-27:
> And the bram mapper could recognize this and map it directly to an asymmetric bram primitive, without having to change each and every consumer of `$mem*` to know about asymmetric brams.
 
Thanks for explaining. This is a convincing argument. Detecting specific uses of mask bits in `memory_bram` is far easier than changing `$mem` cells, and emitting this kind of pattern directly from a frontend is also not complex at all.

> We've been discussing single-port RAMs recently and the path forward seems to be to either extend memory_bram, or write a replacement, but not change the `$mem*` cells.

I was thinking about single-port RAM too, and came to the conclusion that no changes to cells could possibly be desirable, because the only thing that is needed is to detect that the address for the read port and write port is the same. Actually, I would expect that the SAT-based sharing pass would already wire the same net to both ports in such cases, so the detection logic in `memory_bram` would be trivial. Is that not the case?

cliffordwolf, 2019-06-27:
The current sat-based mem-port sharing pass will never consider sharing a read port and a write port. But we would do pretty much the exact same thing to find single-port memories.

One idea for the path forward is to make a survey of the bram primitive features in different FPGA architectures, and then write a pragmatic bram mapper that supports the supersets of all of those features, but no weird extremes with no practical relevance, such as memories with 8 write ports. (The current `memory_bram` pass lacks some important features required to fully utilize some real-world BRAMs, but is also over-engineered wrt some other features.)

daveshah1, 2019-06-27:
> One idea for the path forward is to make a survey of the bram primitive features in different FPGA architectures, and then write a pragmatic bram mapper that supports the supersets of all of those features

I started looking at this, I've gone through iCE40 and ECP5 but still need to look at the various Xilinx families: https://docs.google.com/document/d/1SEWz0d95PAVvl60tDudrdNmDgbCidiGwCLzu-UeN86M/edit?usp=sharing

whitequark, 2019-06-27:
See also this nMigen issue: https://github.com/m-labs/nmigen/issues/12

mithro, 2019-06-27:
@HackerFoo - FYI

PR #3189 (2022-05-18): Add memory_libmap pass.
Fixes #1959.
Fixes #1134.
Fixes #1958.
Fixes #1836.
Fixes #1101.
Fixes #1087.
Fixes #1088.
Fixes #1802.
Fixes #1550.
Fixes #3205.
closed 2022-05-18 by mwkmwkmwk