YosysHQ/yosys #1135: Over-aggressive pmux2shiftx inferring $sub

eddiehung, 2019-06-26:
## Steps to reproduce the issue

./yosys -p "proc; pmux2shiftx; opt -full; stat pmux2shiftx.v`

```
module top(input [7:0] i, output o);
always @*
case (i[6:3])
    4: o <= i[0];
    3: o <= i[2];
    7: o <= i[3];
    default: o <= 1'b0;
endcase
endmodule
```

Creates:
```
   Number of cells:                  4
     $and                            1
     $shift                          1
     $shiftx                         1
     $sub                            1
```
which includes a `$sub` cell (due to offset correction). This `$sub` cell can get inferred into a tech-specific carry chain from which it can't get optimised away.
[image]

## Expected behavior

Ideally, ability to create a better `$shiftx` that can be created that will eventually boil down to a 6-input LUT?
`yosys -p "proc; pmux2shiftx; opt -full; techmap; abc -lut 6; stat" pmux2shiftx.v` shows that this is possible.

More concretely: 
`./yosys -p "proc; synth_xilinx -nosrl" pmux2shiftx.v` returns a LUT6 (`-nosrl` flag is necessary otherwise a `pmuxshiftx` is run internally)
but:
`./yosys -p "proc; pmux2shiftx; synth_xilinx -nosrl" pmux2shiftx.v` returns a bigger circuit with multiple LUTs and carry elements.

Otherwise, perhaps option to control `pmux2shiftx`to not perform such conversions?

## Actual behavior

`$sub` cell that gets mapped to a carry chain.

cliffordwolf, 2019-06-26:
Clearly `pmux2shiftx` needs a better heuristic for when to infer that `$sub` cell.

The current heuristic is a following: If the density (percentage of non-x inputs for the shiftx) is over the min-density (default: 50) without the `$sub` ("absolute density") then don't infer the `$sub`. If absolute density is below that threshold, but density with `$sub` ("range density") is above it, then infer the `$sub`. Otherwise don't transform to shiftx.

In this example "absolute density" is 37%, "range density" is 60%, and min density is at the default value of 50%. Thus we get the `$sub` solution.

Using for example `pmux2shiftx -min_density 30` will yield the non-`$sub` solution.

I would argue that the issue here is not the default value of `-min_density`, but the heuristic using it.

I'm open for suggestions for better heuristics.

Here is where we decide whether to use the solution involving a $sub cell, the solution without $sub cell, or not transform at all: https://github.com/YosysHQ/yosys/blob/master/passes/opt/pmux2shiftx.cc#L657-L676

eddiehung, 2019-06-27:
Detailed explanation, thanks. I don't have a suggestion for a better heuristic right now...

However, a potential enhancement would be to ask for similar behaviour to `muxcover -nodecode` in that I only want to accept zero/low-cost transformations -- i.e. ones that don't require a `$sub` (but perhaps we could continue to accept XOR masking). I'm debugging a regression that seems to increase the number of carry chains in a benchmark because we are running `pmux2shiftx` early and opportunistically in case we identify dynamic-length shift registers (and wide muxes) even if we end up finding none downstream. The `$sub` gets techmapped into a carry chain, and then that chain does not get optimised away in the same way a soft-logic implementation would.

cliffordwolf, 2019-06-27:
> However, a potential enhancement would be to ask for similar behaviour to `muxcover -nodecode`

#1143 adds `pmux2shiftx -norange`.

PR #1143 (2019-06-27): Add "pmux2shiftx -norange"
Fixes #1135
commit 7c14678e (2019-06-27): Add "pmux2shiftx -norange", fixes #1135
closed 2019-06-27 by eddiehung