YosysHQ/yosys #1144: Fails to parse edge sensitive path in specify

eddiehung, 2019-06-27:
## Steps to reproduce the issue

```
module top(input clk, d, output q);
specify
  // Fails:
  (posedge clk => (q +: d)) = (3,1);
  (/*posedge*/ clk => (q +: d)) = (3,1);
  (posedge clk *> (q +: d)) = (3,1);
  (/*posedge*/ clk *> (q +: d)) = (3,1);

  // Works:
  (/*posedge*/ clk => q) = (3,1);
  (/*posedge*/ clk *> q) = (3,1);
endspecify
endmodule
```

From: http://verilog.renerta.com/source/vrg00014.htm

cliffordwolf, 2019-06-28:
Note: This is in non-specify mode. `read_verilog -specify` can parse this just fine.

cliffordwolf, 2019-06-28:
> `(/*posedge*/ clk => (q +: d)) = (3,1);`
> `(/*posedge*/ clk *> (q +: d)) = (3,1);`

That's not valid specify syntax afaict. `(q +: d)` requires an edge. #1147 will fix fix the rest.

(Yes, `read_verilog -specify` parses this invalid syntax. I was more concerned about parsing legal code than rejecting illegal code when I wrote `read_verilog -specify`.)

PR #1147 (2019-07-03): Improve specify dummy parser
Fixes #1144
commit af744097 (2019-07-03): Improve specify dummy parser, fixes #1144
closed 2019-07-03 by cliffordwolf