YosysHQ/yosys #1157: RFC: Ignore SIGINT in interactive mode

bwidawsk, 2019-07-03:
Since the yosys interpreter behaves a lot like a shell, and many people have a habit of using ctrl-c to abort the current command, it seems to be a common occurrence for accidentally aborting the interpreter and losing work.

This RFC proposes that SIGINT is blocked by yosys, perhaps state-fully (in that a second ctrl-c will abort). Instead ctrl-d, or the exit command would be required (or SIGKILL) to end the interpreter.

eddiehung, 2019-07-09:
A very reasonable request @bwidawsk! 
In Python, Ctrl+C is caught, but Ctrl+D is what actually exits. Would you be up for giving it a go and PR it?

bwidawsk, 2019-07-09:
Yeah, I'm happy to give it a shot.

closed 2023-01-21 by bwidawsk