YosysHQ/yosys #1178: Different results with and without debug flag "-g"

eddiehung, 2019-07-09:
## Steps to reproduce the issue

```
./yosys -p "synth_xilinx -abc9" picorv32.v &> nodebug.out
./yosys -p "echo on; synth_xilinx -abc9" picorv32.v -g &> debug.out
diff debug.out nodebug.out -y
```
gives
```
=== design hierarchy ===                                        === design hierarchy ===

   picorv32_wb                       1                             picorv32_wb                       1
     $paramod$4d2dfdcc1db1a7362453fb449ccdda75bb1b39f9\picorv        $paramod$4d2dfdcc1db1a7362453fb449ccdda75bb1b39f9\picorv

   Number of wires:               1175                        |    Number of wires:               1170
   Number of wire bits:           4107                        |    Number of wire bits:           4102
   Number of public wires:         181                             Number of public wires:         181
   Number of public wire bits:    1760                             Number of public wire bits:    1760
   Number of memories:               0                             Number of memories:               0
   Number of memory bits:            0                             Number of memory bits:            0
   Number of processes:              0                             Number of processes:              0
   Number of cells:               2351                        |    Number of cells:               2346
     CARRY4                        101                               CARRY4                        101
     FDRE                          680                               FDRE                          680
     LUT1                            9                               LUT1                            9
     LUT2                          702                        |      LUT2                          703
     LUT3                          110                        |      LUT3                           99
     LUT4                          138                        |      LUT4                          132
     LUT5                          439                        |      LUT5                          443
     LUT6                          107                        |      LUT6                          115
     MUXF7                           1                        <
     RAM32X1D                       64                               RAM32X1D                       64

   Estimated number of LCs:       1021                        |    Estimated number of LCs:       1025
```

## Expected behavior

No divergence.

## Actual behavior

Divergence.

eddiehung, 2019-07-09:
First point of divergence appears to be in the `coarse` stage of `synth_xilinx`.. specifically within its `wreduce` call (a `stat` before this call is identical):
```
./yosys -p "synth_xilinx -abc9 -run :coarse; proc; opt_expr; opt_clean; check; opt; wreduce; stat" picorv32.v &> nodebug.out
./yosys -p "synth_xilinx -abc9 -run :coarse; proc; opt_expr; opt_clean; check; opt; wreduce; stat" picorv32.v -g &> debug.out
diff debug.out nodebug.out -y
```
```
=== design hierarchy ===                                        === design hierarchy ===

   picorv32_wb                       1                             picorv32_wb                       1
     $paramod$4d2dfdcc1db1a7362453fb449ccdda75bb1b39f9\picorv        $paramod$4d2dfdcc1db1a7362453fb449ccdda75bb1b39f9\picorv

   Number of wires:                700                        |    Number of wires:                699
   Number of wire bits:           5379                        |    Number of wire bits:           5371
   Number of public wires:         197                             Number of public wires:         197
```

eddiehung, 2019-07-09:
Next divergence here: https://github.com/YosysHQ/yosys/blob/27b27b8781ab8d57aa85a432aba7e914570feffb/techlibs/xilinx/synth_xilinx.cc#L284

```
./yosys -p "synth_xilinx -abc9 -run :fine; opt -fast -full; memory_map; dffsr2dff; dff2dffe; opt -full; techmap -map +/techmap.v -map +/xilinx/arith_map.v; opt -fast; stat" picorv32.v &> nodebug.out
./yosys -p "synth_xilinx -abc9 -run :fine; opt -fast -full; memory_map; dffsr2dff; dff2dffe; opt -full; techmap -map +/techmap.v -map +/xilinx/arith_map.v; opt -fast; stat" picorv32.v -g &> debug.out
diff debug.out nodebug.out -y
```
gives
```
=== design hierarchy ===                                        === design hierarchy ===

   picorv32_wb                       1                             picorv32_wb                       1
     $paramod$4d2dfdcc1db1a7362453fb449ccdda75bb1b39f9\picorv        $paramod$4d2dfdcc1db1a7362453fb449ccdda75bb1b39f9\picorv

   Number of wires:               1370                        |    Number of wires:               1368
   Number of wire bits:           7213                        |    Number of wire bits:           7204
   Number of public wires:         194                             Number of public wires:         194
```

cliffordwolf, 2019-07-15:
0e6c830 fixes the behavior for this test case (just pushed to your eddie/fix1178 branch), but does not completely fix the underlying issue. Here is what's happening:

Calling `log_id()` on an object will cause this ID to be added to `log_id_cache` to prevent the memory holding the string to be free'd before it's printed. Printing a new header will automatically clear out `log_id_cache`.

Turning debug on/off toggles if certain object names are printed before the object is removed. So printing the name will then delay the removal of the ID until the next header is print, but silently printing the name will free the ID right away. Therefore, if a new ID is created in that time period, that new ID will get assigned a different integer, depending on if the ID for the name of the removed object was printed and thus is still hanging around. This changes which names are assigned which numeric IDs.

Thus, when a later pass does something in sort order of a bunch of IdString objects, that sort order may then change because IdStrings are simply sorted by their numeric ID.

As a quick fix I've now added a `log_checkpoint()` command that triggers clearing out `log_id_cache` right away.

The proper way to fix this is probably to manage a to-be-deleted list in IdString, and whenever the ref counter of an ID counts down to zero, instead of removing the ID right away, we just add it to a to-be-deleted list, and then process this list and actually remove IdStrings when we print the next header. This would then replace the `log_id_cache` mechanism and `log_id()` will no longer influence the order in which IDs are freed and allocated.

cliffordwolf, 2019-07-15:
> The proper way to fix this is probably [...]

Two better ideas:

(1) `log_id()` could simply create a copy of the string, and `log_id_cache` would become a container of strings instead of a container of IdStrings.

(2) Free the ID when the refcount goes to zero, but the string is moved to a special place and only de-allocated when the next header is printed.

I think I prefer (1) because it's fairly simple and performance shouldn't be impacted (in the non-debug case) because if we print so many IDs that creating those str copies becomes a performance issue, then we should probably make those log messaged debug log messages anyway.

cliffordwolf, 2019-07-15:
> I think I prefer (1) [...]

I've now implemented this in 44fd459 and pushed it to the `eddie/fix1178` branch.

eddiehung, 2019-07-15:
Ughhhh I can't repro this on master @ a97d30d2 right now

eddiehung, 2019-07-15:
Okay the `eddie/fix1178` seems to fix it for me on this testcase, with and without `-abc9`.

Will create a PR.

commit 44fd459c (2019-07-15): Redesign log_id_cache so that it doesn't keep IdString instances referenced, fixes #1178
PR #1196 (2019-07-15): Fix different synth results between with and without debug output "-g"
Fixes #1178
closed 2019-07-15 by eddiehung