YosysHQ/yosys #1192: support cross-module task invocation?

andres-erbsen, 2019-07-13:
Hi,

I am wondering how much work it would take to support invocation of tasks from outside of the module that defines them. Here is a minimal example:

```verilog
module counter;
  reg [31:0] cnt = 0;
  task increment; begin
    cnt = cnt + 1;
  end endtask
endmodule

module top(input wire clk);
  counter C();
  always @(posedge clk) begin
    C.increment(); // yosys: ERROR: Can't resolve task name `\C.increment'.
  end
endmodule
```

[Internets](https://stackoverflow.com/questions/37624934/how-to-call-tasks-from-a-separate-module-in-verilog) tell me that this is not supposed to work, but it does in `iverilog`, and I would like to use it in yosys for prototyping in the context of a larger system.

I would be be happy to try my hand at implementing this, but first I am reaching out to hear whether there are any gotchas that would make this much more work than replacing code like https://github.com/YosysHQ/yosys/blob/1706798f4e595266a8758ae6d0ff9d978299de10/frontends/ast/simplify.cc#L2231 with... something recursive.

[Here](https://gist.github.com/andres-erbsen/7ae70c455eca1b0da3ced0636e94f6a2) is a more complete example of the desired behavior.

Thanks,
Andres

AaronKel, 2019-07-16:
Please note iverilog is a simulator, yosys is a synthesis engine. All things that can synthesize can simulate but not vice versa. You have a few problems with your code but the biggest problem with this code however is that there is no ports to your task, so synthesis tools would just black box these as logic abstractions. It doesn't have a sense of what drives cnt. It just sees a task with no ports so would therefore not be able to be synthesizable unfortunately.

Looking at your complete example it looks more like a testbench than a digital implimentation, try to think of how you would implement this as digital logic.

My question is why is it coded in such a manner? For abstraction?

andres-erbsen, 2019-07-17:
I am aware of the difference between simulation and synthesis, and that I posted an illustrative testbench not an implementation-ready design, and the latter would need to include ports to the outside world. What I am doing here *is* an experiment in verilog coding style, quite far from accepted practice and hence reasonably deserving of skepticism, but issue is not in me being unable to implement working designs in verilog. And right now I am looking to try out this experiment on something bigger than conveniently written in a single file/module, hence this issue.

So, devs, is there anything that I should look into or keep in mind wrt cross-module task invocation, or should I just dive in and try to implement it?

daveshah1, 2019-07-17:
Like many synthesis tools Yosys elaborates each Verilog module to its own netlist module, the only interface between the two being input and output pins. This can optionally be flattened later in the flow but flattening is an optional transformation.

As a result its not just a case of "reaching" into the other module during RTLIL from AST generation, but something a bit more complicated. Probably what you'd need to do is store the task call in the calling module, then during `hierarchy` rebuild the called module with task "triggers" exposed as additional inputs (in the same way that specialised versions of parameterisable modules instantiated with non-default parameters are created here).

This would certainly be a non-trivial change, and I'm not sure if it would even be accepted upstream (given the significant disruption to support a rarely-used, non-synthesisable feature).

andres-erbsen, 2019-07-17:
Plenty thanks for the thorough answer! What you describe makes sense, although I was really hoping I would hear something different.

I am now thinking I will make myself a hacky source-level module inliner or a namespacing preprocessor instead until I have actually convinced myself that what I am trying to do can scale and synthesize well.

closed 2019-07-17 by andres-erbsen

udif, 2019-07-17:
I think you should ask yourself what are you trying to achieve in this coding style, and what advantage do you think it has over more conventional synthesizable coding tyles.