YosysHQ/yosys #1201: cmp2lut fails on LUT6

eddiehung, 2019-07-16:
## Steps to reproduce the issue

I've extended cmp2lut tests to include LUT6 instead of just LUT4, and they seem to fail equivalence now: https://github.com/YosysHQ/yosys/compare/cmp2lut_lut6

The origin of this is that the following line (copied from `synth -run coarse` invocation in the past, but adapted to be called with `-D LUT_WIDTH=6` instead of the default `-D LUT_WIDTH=0`)
https://github.com/YosysHQ/yosys/blob/dd10d2b00d6760f2d09f55c90591855afcae409e/techlibs/xilinx/synth_xilinx.cc#L277

## Expected behavior

Equivalence passing.
yosys-tests' bigsim tests on `navre_xilinx*` to now work.

## Actual behavior

Equivalence fails:
```
4.5. Executing EQUIV_STATUS pass.
Found 22 $equiv cells in equiv:
  Of those cells 10 are proven and 12 are unproven.
  Unproven $equiv $auto$equiv_make.cc:251:find_same_wires$209: \o4_4_gold \o4_4_gate
  Unproven $equiv $auto$equiv_make.cc:251:find_same_wires$208: \o4_3_gold \o4_3_gate
  Unproven $equiv $auto$equiv_make.cc:251:find_same_wires$201: \o3_2_gold \o3_2_gate
  Unproven $equiv $auto$equiv_make.cc:251:find_same_wires$200: \o3_1_gold \o3_1_gate
  Unproven $equiv $auto$equiv_make.cc:251:find_same_wires$197: \o2_4_gold \o2_4_gate
  Unproven $equiv $auto$equiv_make.cc:251:find_same_wires$196: \o2_3_gold \o2_3_gate
  Unproven $equiv $auto$equiv_make.cc:251:find_same_wires$195: \o2_2_gold \o2_2_gate
  Unproven $equiv $auto$equiv_make.cc:251:find_same_wires$194: \o2_1_gold \o2_1_gate
  Unproven $equiv $auto$equiv_make.cc:251:find_same_wires$191: \o1_4_gold \o1_4_gate
  Unproven $equiv $auto$equiv_make.cc:251:find_same_wires$190: \o1_3_gold \o1_3_gate
  Unproven $equiv $auto$equiv_make.cc:251:find_same_wires$189: \o1_2_gold \o1_2_gate
  Unproven $equiv $auto$equiv_make.cc:251:find_same_wires$188: \o1_1_gold \o1_1_gate
Found a total of 12 unproven $equiv cells.
ERROR: Found 12 unproven $equiv cells in 'equiv_status -assert'.
```
yosys-tests' bigsim tests on `navre_xilinx*` failing (but passes when I comment out the cmp2lut)

eddiehung, 2019-07-16:
CC @whitequark who authored cmp2lut

whitequark, 2019-07-16:
@eddiehung I suspect the problem here is Verilog implicit integer widths, since 1<<4=16 and 1<<6=64; this transitions through the default width 32. I wouldn't be able to immediately give this issue attention unfortunately, so feel free to investigate that direction, or leave it to me to fix it a bit later.

eddiehung, 2019-07-16:
Thanks for the background info, which I'll use to see if I can fix it, otherwise I'll temporarily remove it from `synth_xilinx` and assign this issue to you to work on when you're available.

whitequark, 2019-07-16:
@eddiehung I think you forgot to commit `check_map_lut6.ys`

eddiehung, 2019-07-16:
Sorry, and thanks. I think I solved it -- you are returning a 32-bit integer from the function `gen_lut`. PR incoming.

PR #1202 (2019-07-16): cmp2lut transformation to support >32 bit LUT masks
Fixes #1201
closed 2019-07-16 by eddiehung