YosysHQ/yosys #1210: Extract fir file from a verilog project

mlulaj, 2019-07-19:
## Issue
Hello,
I am trying to export firrtl from a verilog project. 

Im using the command write_firrtl and it executes without error, but when I try to use the generated .fir file, to another tool, which takes as input only .fir files, it can not parse it .

Maybe Im doing some step wrong on generating the fir file :

The project which I work, has several files and the top level.  I created a script like this : 

`read_verilog-formal /home/mustafa/Downloads/yosys/tests/Aes128/half_adder.v `
`read_verilog -formal /home/mustafa/Downloads/yosys/tests/Aes128/full_adder.v`
`read_verilog -formal /home/mustafa/Downloads/yosys/tests/Aes128/ripple_adder_4bit.v`
`read_verilog -formal /home/mustafa/Downloads/yosys/tests/Aes128/ripple_adder_6bit.v`
`read_verilog -formal /home/mustafa/Downloads/yosys/tests/Aes128/ripple_adder_8bit.v`
`read_verilog -formal /home/mustafa/Downloads/yosys/tests/Aes128/ripple_adder_12bit.v`
`read_verilog -formal /home/mustafa/Downloads/yosys/tests/Aes128/vedic2x2.v`
`read_verilog -formal /home/mustafa/Downloads/yosys/tests/Aes128/vedic4x4.v`
`read_verilog -formal /home/mustafa/Downloads/yosys/tests/Aes128/vedic8x8.v`
`hierarchy -top vedic8x8`
`proc`
`opt`
`write_firrtl vedic8x8r.fir `

So on this case the top level module is vedi8x8.v
Am I missing something  by doing like this ?
Thank you

daveshah1, 2019-07-19:
What tool are you using it with? What error is it giving? I don't think anyone can help you without that information.

mlulaj, 2019-07-19:
@daveshah1  Its this on : https://github.com/ekiwi/rfuzz   RFUZZ
I tried it with several fir files and it works correctly.

I think the problem is that Im using Yosys wrong. Is it the right way to elaborate and link together modules , when having multiple modules on multiple files ? 
I also tried it with a simple full adder, so basically two files : half_adder.v and full_adder.v
And still can not use the output to the other tool. 

Some of the errors from the RFUZZ : 

`[error] (run-main-0) firrtl.passes.PassExceptions:` 
`[error] firrtl.passes.CheckInitialization$RefNotInitializedException:  @[:full_adder.fir@9.4] : [module full_adder]  Reference _or__home_mustafa_Downloads_yosys_tests_Aes128_full_adder_v_7_3_Y is not fully initialized.`
`[error]    : _or__home_mustafa_Downloads_yosys_tests_Aes128_full_adder_v_7_3_Y <= VOID`
`[error] firrtl.passes.CheckInitialization$RefNotInitializedException:  @[:full_adder.fir@14.4] : [module full_adder]  Reference _0 is not fully initialized.`
`[error]    : _or__home_mustafa_Downloads_yosys_tests_Aes128_full_adder_v_7_3_Y <= VOID`
`[error]    : _0 <= _or__home_mustafa_Downloads_yosys_tests_Aes128_full_adder_v_7_3_Y`
`[error] firrtl.passes.CheckInitialization$RefNotInitializedException:  @[:full_adder.fir@3.4] : [module full_adder]  Reference cout is not fully initialized.`
`[error]    : cout <= VOID`

cliffordwolf, 2019-07-19:
Please provide a [MCVE](https://stackoverflow.com/help/minimal-reproducible-example). If it's not possible to reproduce what you are seeing in a matter of a few minutes max, nobody will bother having a look at the problem.

ekiwi, 2019-07-22:
This might be a problem in the `rfuzz` tool that I authored.
We will debug this in https://github.com/ekiwi/rfuzz/issues/2.
I think you could close this issue for now if @mlulaj agrees.

closed 2019-07-23 by mlulaj