YosysHQ/yosys #1221: Automatically Infer latch

kunalg123, 2019-07-24:
Hi 
For the netlist attached, I need to always use the below test_map.v and add the lines below the below script. Is there a way, latch can be automatically inferred provided we have a LATCH cell in standard cell library? In this case, OSU018 std cells has LATCH 

**test_map.v**
module \$_DLATCH_P_ (input E, input D, output Q);
           LATCH _TECHMAP_REPLACE_ (
               .CLK(E),
               .D(D),
               .Q(Q)
           );
        endmodule
~                                               

# High-level synthesis
synth -top MakeClock
techmap -map test_map.v
simplemap

[MakeClock.v.zip](https://github.com/YosysHQ/yosys/files/3427620/MakeClock.v.zip)

daveshah1, 2019-07-29:
I am not sure what a latch looks like in a liberty file, but this seems like a reasonable potential enhancement to `dfflibmap`. For now the techmap route you are using already is the best option.

LudwigCRON, 2019-12-11:
Hello,

Not sure if it can help. In liberty file the structure I found in many for latches is the following:

```lib
cell(DLATCH_P) {
   area: xxxx;
   cell_description: "high active transparent D-latch";
   latch(IQ, IQN) {
        data_in : D;
        enable: G;
   }
   pin (D) {
        direction: input;
        max_transition: xxx;
        capacitance: ccc;
        timing () {
            ... // setup and hold with respect to G (the gate)
        }
   }
   pin (G) {
        direction: input;
        clock: true;
        max_transition: xxx;
        capacitance: ccc;
        timing () {
           ... // min_pulse_width
        }
   }
   pin (Q) {
       direction: output;
       function: "IQ";
       max_capacitance: ccc;
       max_fanout: n;
       max_transition: ttt;
       timing () {
           ... // rising edge / fall edge with respect to D and G
       }
   }
  // [optionaly pin (QN) same as pin (Q) with function: "IQN"
}
```