YosysHQ/yosys #1223: internal cell error $_DFF_PP1

matthuszagh, 2019-07-25:
```
ERROR: Found error in internal cell \top.$auto$simplemap.cc:496:simplemap_adff$154 ($_DFF_PP1) at kernel/rtlil.cc:1263:
  attribute \module_not_derived 1
  attribute \src "top.v:27|/nix/store/q5p447sz6iph570nzlm7wgngl58hh1zc-yosys-master/bin/../share/yosys/xilinx/ff_map.v:43"
  cell $_DFF_PP1 $auto$simplemap.cc:496:simplemap_adff$154
    connect \R $techmap$techmap165$auto$simplemap.cc:496:simplemap_adff$154.$not$/nix/store/q5p447sz6iph570nzlm7wgngl58hh1zc-yosys-master/bin/../share/yosys/xilinx/ff_map.v:43$110_Y
    connect \C \clk_i
    connect \Q \oe_n_o
    connect \D $0\oe_n_o[0:0]
  end
```

Steps to reproduce:
```
`default_nettype none

module top
        (
         inout wire [7:0] data_io,
         output reg [7:0] rdata_o,
         input wire [7:0] wdata_i,
         input wire       rxf_n_i,
         input wire       txe_n_i,
         output reg       rd_n_o,
         output reg       wr_n_o,
         output reg       siwua_n_o,
         input wire       clk_i,
         output reg       oe_n_o,
         input wire       suspend_n_i
         );

        initial begin
                rdata_o = {8{1'b0}};
                rd_n_o = 1;
                wr_n_o = 1;
                siwua_n_o = 1; /* Never flush TX data. */
        end

        assign data_io = (!txe_n_i && !wr_n_o) ? wdata_i : {8{1'bz}};

        always @(posedge clk_i, negedge suspend_n_i) begin
                if (!suspend_n_i) begin
                        wr_n_o <= 1;
                        rd_n_o <= 1;
                        oe_n_o <= 1;
                        rdata_o <= {8{1'b0}};
                end
                else begin
                        // Give TX bus precedence.
                        if (!txe_n_i) begin
                                wr_n_o <= 0;
                                rd_n_o <= 1;
                                oe_n_o <= 1;
                                rdata_o <= {8{1'b0}};
                        end
                        // oe_n_o must be driven low at least one period before we can read.
                        else if (!rxf_n_i && oe_n_o) begin
                                wr_n_o <= 1;
                                rd_n_o <= 1;
                                oe_n_o <= 0;
                                rdata_o <= {8{1'b0}};
                        end
                        else if (!rxf_n_i && !oe_n_o) begin
                                wr_n_o <= 1;
                                rd_n_o <= 0;
                                oe_n_o <= 0;
                                rdata_o <= data_io;
                        end
                        else begin
                                wr_n_o <= 1;
                                rd_n_o <= 1;
                                oe_n_o <= 1;
                                rdata_o <= {8{1'b0}};
                        end
                end // else: !if(!suspend_n_i)
        end

endmodule // usb
```
`yosys top.v -p "synth_xilinx"`

daveshah1, 2019-07-25:
Thanks for reporting! #1224 should fix the internal cell error.

While looking into this, I also found that the tristate support for Xilinx isn't working at the moment, for this I have created #1225. In the mean time, you would need to manually instantiate an IOBUF.

closed 2019-07-25 by eddiehung

matthuszagh, 2019-07-25:
I'm still getting this error
```
ERROR: Found error in internal cell \usb.$auto$simplemap.cc:496:simplemap_adff$292774 ($_DFF_PP1) at kernel/rtlil.cc:1263:
  attribute \module_not_derived 1
  attribute \src "hdl/usb.v:29|/nix/store/kqrkq4rzx533wm25458qrhfqd1r5hv3i-yosys-master/bin/../share/yosys/xilinx/ff_map.v:43"
  cell $_DFF_PP1 $auto$simplemap.cc:496:simplemap_adff$292774
    connect \C \clk_i
    connect \D $0\oe_n_o[0:0]
    connect \Q \oe_n_o
    connect \R $techmap$techmap292778$auto$simplemap.cc:496:simplemap_adff$292774.$not$/nix/store/kqrkq4rzx533wm25458qrhfqd1r5hv3i-yosys-master/bin/../share/yosys/xilinx/ff_map.v:43$5784_Y
```
Does it not occur when you run the example?

matthuszagh, 2019-07-25:
It seems weird that it's complaining about the cell without a terminal underscore. I grepped the source code and all instances of that cell were`_DFF_PP1_`.

daveshah1, 2019-07-25:
No, I don't see the error any more. Can you run `cat /nix/store/kqrkq4rzx533wm25458qrhfqd1r5hv3i-yosys-master/bin/../share/yosys/xilinx/ff_map.v` and check it has the terminal underscore?

matthuszagh, 2019-07-25:
My mistake, it was an issue with the way I was using Nix. Your fix works! thx

PR #1224 (2019-07-25): xilinx: Fix missing cell name underscore in cells_map.v
Fixes #1223