YosysHQ/yosys #1229: Fundamental combinatorial synthesis error. . . ABC script problem?

RTimothyEdwards, 2019-07-26:
I spent half of yesterday trying to figure out why my version of the picosoc simulated differently between the original verilog source and the synthesized verilog netlist.  I had made numerous changes in qflow, so I was convinced it was something I did.  However, in the end, I found that a combinational expression had been incorrectly optimized, presumably by ABC.  The mem_16bit_buffer[15] bit became undefined while mem_state was 2'b01.  This absolutely should not happen, as mem_16bit_buffer should be untouched when mem_state is 1.  It appears that the undefined state of the RAM at that point was forcing the next state of that single bit to be undefined, when it should have been a don't-care condition.

The only thing I can find that I did differently on this synthesis run is that I adapted an ABC synthesis script that I got from Mohamed Shalan (which he uses in CloudV).  This could be a very complicated debug, so before I trigger that, I would like to post the ABC script I use, and I would like to know if there is something I did that is obviously wrong with it.  However, I tend to think that this is a real bug in ABC.

The original ABC script is this:

    `+strash;scorr;ifraig;retime,{D};strash;dch,-f;map,-M,1,{D}`

The ABC script that showed the bug is this:

    `+fx;mfs;strash;refactor;balance;rewrite;refactor;balance;rewrite;rewrite,-z;balance;refactor,-z;rewrite,-z;balance;retime,-D,{D},-M,5;scleanup;map,-M,0;retime,-D,{D};buffer,-N,12,-S,230;upsize,{D};dnsize,{D};stime,-p;`

The main purpose of the modified script was to force ABC to keep all fanouts less than 12 so that there would be no need for my fanout-handling tool in qflow to do anything except create the clock buffer tree.

The "upsize" and "dnsize" are probably unnecessary because the qflow gate resizing script should take care of the timing, but I was interested in knowing what result ABC would generate from gate resizing.

I have not (yet) tried any extensive experiments in modifying the ABC script to see if I can isolate the problem to a single statement.  I wanted to raise the issue first and get an opinion from somebody who knows more about the ABC scripts than I do to see (1) if I did something obviously bad, and (2) if any of the statements in the script are known to be "experimental" and less well vetted than others.

The synthesis is in X-Fab XH018, so I cannot post the netlist here.  If this seems like an issue that would raise alarms, then I can try some experiments to see if I can reproduce the error in an open technology.  I'm doubtful that the error is going to be easily reproduced, though.

---Tim

udif, 2019-07-26:
I am not familiar with the picosoc codebase, so I cannot analyze your issue, but gate level vs RTL simulation mismatch is not uncommon, and not necessarily due to synthesis bugs.  
From what I can learn from your description, you haven't actually pinpointed a specific synthesys error, you merely witnessed a signal getting turne into X when it was supposed to keep its value in RTL. 
In that case, it is possible that your mismatch is not due to synthesis errors, but due to X pessimism, where an expression should have a known output regardless of the input state, but when that input is X, the resulting output is X.

From a quick google lookup, here is good discussion of X pessimism:
https://www.gaisler.com/doc/antn/GRLIB-AN-0010.pdf

Until you actually extract the logic cone causing the X output and study it, it's hard to tell the source of the error.

daveshah1, 2019-07-28:
This is not an actionable issue without the following:

 - Input Verilog source (how exactly is picosoc set up), liberty file, full Yosys script and testbench
 - the problematic synthesised netlist
 - ideally, a screenshot from a waveform viewer of the issue, and a line reference to the problematic line(s) in the output netlist if you can find them

RTimothyEdwards, 2019-07-28:
Understood that this is an incomplete entry.  I was thinking of trying to run it with an open standard cell set like OSU018 although I'm doubtful that it will be reproducible that way, given that only one net was in error in an entire SoC, and the error could be related to mapping and so specific to the standard cell set used, which was D_CELLS in X-Fab XH018 (also a slightly outdated D_CELLS because the latest one is on a tighter route pitch and I can't get qrouter to route it without lots of DRC errors).  Obviously, the X-Fab liberty file is proprietary so I can't just post it here.  Everything else, I can provide;  the problem is that without the liberty file, you won't be able to reproduce the error, and without the verilog for the standard cells, you won't be able to simulate the error.  One way to reproduce the problem would be to do it on the efabless platform.  I could share the project with you and point to exactly how to produce the error.

I started drawing out the combinatorial logic that is generating the error, but I have to check each standard cell's function, because the names are obscure.  There is always a chance (unlikely, in my opinion) that it is, as mentioned above, an "X pessimism" problem (unlikely because I've never seen it happen in years of working with the X-Fab XH018 D_CELLS library).  One thing I am certain of is that it is not a race condition or timing problem, because the signal that should have a clean binary value is undefined for an entire clock cycle.  The problem is clearly in the combinatorial logic.

Let me know how you would like me to provide you with a reproducible example.

Thanks,
Tim

RTimothyEdwards, 2019-07-30:
I am going to close this issue because I am unable to reproduce it outside of the specific conditions on the efabless platform.  With all files moved to my local desktop, and all conditions and files the same except for the revision of yosys, the error did not repeat.  Sorry for the bother.  I just wanted to check for obvious things before I went to the trouble of collecting everything together for a reproducible case.

closed 2019-07-30 by RTimothyEdwards