YosysHQ/yosys #1230: Issues trying to compile latest (MacOS base)

lvcargnini, 2019-07-26:
## Steps to reproduce the issue
- Just pulled the latest github source code
- using a macos, latest version 10.14.6
- Latest Xcode
- latest llvm using brew
installed the packages "open /Library/Developer/CommandLineTools/Packages/macOS_SDK_headers_for_macOS_10.14.pkg", so it finds the wchar.h

- have it configured from previous compilations,
- brew install bison flex gawk libffi git graphviz pkg-config python3
all installed and set 

## Expected behavior
compile properly no errors
## Actual behavior
compilation errors:

>  13%] Building frontends/json/jsonparse.o
> [ 13%] Building frontends/liberty/liberty.o
> [ 14%] Building frontends/verific/verific.o
> [ 14%] Building frontends/verific/verificsva.o
> In file included from frontends/verific/verific.cc:32:
> ./frontends/verific/verific.h:22:10: fatal error: 'DataBase.h' file not found
> #include "DataBase.h"
>          ^~~~~~~~~~~~
> 1 error generated.
> make: *** [frontends/verific/verific.o] Error 1
> make: *** Waiting for unfinished jobs....
> In file included from frontends/verific/verificsva.cc:73:
> ./frontends/verific/verific.h:22:10: fatal error: 'DataBase.h' file not found
> #include "DataBase.h"
>          ^~~~~~~~~~~~
> 1 error generated.
>

daveshah1, 2019-07-26:
It seems like it is trying to find a Verific header. Have you set ENABLE_VERIFIC to 1 in the Makefile? This should be left as 0.

lvcargnini, 2019-08-01:
Yes it was set to 1, just changed it to 0. Passed, but found a new problem:

[ 76%] Building backends/protobuf/protobuf.o
backends/protobuf/protobuf.cc:29:10: fatal error: 'yosys.pb.h' file not found
#include "yosys.pb.h"
         ^~~~~~~~~~~~
1 error generated.
make: *** [backends/protobuf/protobuf.o] Error 1
make: *** Waiting for unfinished jobs....

lvcargnini, 2019-08-01:
I just manually did:
cd misc
protoc --cpp_out "../backends/protobuf" yosys.proto

and it seems to have worked, got the following thou:
[ 99%] Building yosys-filterlib
ld: warning: directory not found for option '-L/usr/local/opt/boost/lib'
[100%] Building yosys
[100%] Building libyosys.so
ld: warning: directory not found for option '-L/usr/local/opt/boost/lib'
ld: warning: directory not found for option '-L/usr/local/opt/boost/lib'

  Build successful.

installing boost and re-building after to see how it goes

lvcargnini, 2019-08-01:
ok, it is necessary to 'brew install boost' to fix that:
 99%] Building share/xilinx/brams_init_32.vh
[ 99%] Building share/xilinx/brams_init_18.vh
[ 99%] Building share/xilinx/brams_init_16.vh
[ 99%] Building share/xilinx/brams_init_9.vh
[ 99%] Building share/xilinx/brams_init_8.vh
[ 99%] Building yosys-filterlib
[100%] Building yosys
[100%] Building libyosys.so

  Build successful.

All set now

closed 2019-08-01 by lvcargnini