YosysHQ/yosys #1243: Unexpected behaviour of for loop and if statement

ymherklotz, 2019-08-01:
## Steps to reproduce the issue

Consider the following piece of code:

```verilog
module top  (y, clk, sel);
   output wire y  ;
   input       clk; 
   input       sel;
   reg         reg_assign = (1'h0) ;
   reg [1:0]   reg_count = (1'h0) ;
   assign y = reg_assign ;
   always @(posedge clk) 
     if (sel)
       for (reg_count = 0; reg_count < 2; reg_count = reg_count + 1'h1)
         if (0);
         else reg_assign <= 1;
     else reg_assign <=  0;
endmodule
```

First of all, the for loop in the code does seem a bit dodgy, however, I would still expect `reg_assign` to be set to 1 when `sel` is high. When `sel` is low, `reg_assign` should then be reset to 0. Removing the dead if statement in the for loop results in the correct behaviour.

However, when synthesised with 

```
Yosys 0.8+618 (git sha1 acd8bc0a, clang  -fPIC -Os)
```

using

```
yosys -p "read_verilog rtl.v; synth; write_verilog -noattr synth.v"
```

`reg_assign` is set to a constant 0 instead of to what the value of `sel` is.

I have also attached a folder containing a test bench and SymbiYosys script to compare the design to the synthesised net list.

## Expected behaviour

I would expect this to be implemented by assigning `sel` to `y`. This is actually also the output of a previous version of Yosys (`Yosys 0.8+508 (git sha1 c2ea3746, clang 8.0.0 -fPIC -Os)`)

```verilog
module top(y, clk, sel);
  input clk;
  wire reg_assign;
  wire [1:0] reg_count;
  input sel;
  output y;
  assign reg_assign = 1'h0;
  assign reg_count = 2'h0;
  assign y = sel;
endmodule
```

## Actual behavior

However, with Yosys, `y` is set to a constant 0.

```verilog
module top(y, clk, sel);
  input clk;
  wire reg_assign;
  wire [1:0] reg_count;
  input sel;
  output y;
  assign reg_assign = 1'h0;
  assign reg_count = 2'h0;
  assign y = 1'h0;
endmodule
```

[test.zip](https://github.com/YosysHQ/yosys/files/3457004/test.zip)

daveshah1, 2019-08-05:
Reverting #1169 seems to give the expected result (I don't think assigning sel to y is correct, but the register looks good):

```verilog
module top(y, clk, sel);
  input clk;
  reg reg_assign = 1'h0;
  wire [1:0] reg_count;
  input sel;
  output y;
  always @(posedge clk)
      reg_assign <= sel;
  assign reg_count = 2'h0;
  assign y = reg_assign;
endmodule
```

cc @whitequark

ymherklotz, 2019-08-05:
Ah yes you are right, assigning `sel` directly does not seem correct, clocking it is definitely right.

whitequark, 2019-08-05:
@daveshah1 Good catch, I'll fix the pass in a moment.

whitequark, 2019-08-08:
Fixed in https://github.com/YosysHQ/yosys/pull/1267.

PR #1267 (2019-08-09): proc_prune: fix handling of exactly identical assigns
Before this commit, in a process like:
```
   process $proc$bug.v:8$3
     assign $foo \bar
     switch \sel
       case 1'1
         assign $foo 1'1
         assign $foo 1'1
       case
         assign $foo 1'0
     end
   end
```
both of the `assign $foo 1'1` would incorrectly be removed.

Fixes #1243.
commit 0b09a347 (2019-08-09): proc_prune: fix handling of exactly identical assigns.

Before this commit, in a process like:
   process $proc$bug.v:8$3
     assign $foo \bar
     switch \sel
       case 1'1
         assign $foo 1'1
         assign $foo 1'1
       case
         assign $foo 1'0
     end
   end
both of the "assign $foo 1'1" would incorrectly be removed.

Fixes #1243.
closed 2019-08-09 by whitequark