YosysHQ/yosys #1244: Add optional step in synth_xilinx to insert BUFG.

koriakin, 2019-08-01:
1. Collect all nets used as clocks.
  1.1. Collect a list of all clock input ports.
  1.2. If a clock input is driven by a mergable inverter, mark the inverter input as clock instead.
2. Skip any nets already driven by a clock buffer primitive.
3. Count already-present BUFG/BUFGMUX/BUFGCTRL/...
4. Set limit to 8 - num (Spartan 6) or 32 - num (series 7)
5. Insert BUFGs on top $limit nets

eddiehung, 2019-08-02:
I would say steps 3-5 are not critical, and I would suggest a first PR without them. Yosys does not currently have resource limit awareness, and I would like to open a separate discussion about how to do this in a more generic fashion -- one that would allow users to specify priority-by-fanout, priority-by-slack, etc.

eddiehung, 2019-08-29:
Can requirements 1-2 considered fixed by #1292?
3-5 should be tackled in another (lower-priority) ticket -- would you agree?

koriakin, 2019-08-29:
I agree.

Also, we don't yet cover 1.2, but the only place where this will currently by a problem is blockram mapping (which can be fixed to use IS_*_INVERTED once we get the extractinv pass, which runs after clkbufmap).

mwkmwkmwk, 2019-12-04:
The inverter issue has been fixed by 6cdea425b81fcfe1eec20cbfc4c4e27d46cb641d, and resource limits are not supported in yosys in general at this point, so I'm closing this issue.

closed 2019-12-04 by mwkmwkmwk