YosysHQ/yosys #1247: dff2dffs: Add option to force S/R value matching INIT value.

koriakin, 2019-08-01:
Make dff2dffs accept a "-match-init" option, with the following semantics:

1. Don't merge synchronous reset into a flip-flop if it has INIT = 1
2. Don't merge synchronous set into a flip-flop if it has INIT = 0
3. Convert any leftover $_DFF_*_ with INIT = 0 into $__DFFS_**0_ with reset tied to 0
4. Convert any leftover $_DFF_*_ with INIT = 1 into $__DFFS_**1_ with set tied to 0

This ensures that tech mapping doesn't need to peek at INIT value to choose the correct output primitive.

eddiehung, 2019-09-14:
Can be closed due to https://github.com/YosysHQ/yosys/pull/1366 being merged?

koriakin, 2019-09-14:
Closed by #1366

koriakin, 2019-09-14:
(only points 1 and 2 done; points 3 and 4 are not necessary anymore since we have _TECHMAP_WIREINIT_)

closed 2019-09-14 by koriakin