YosysHQ/yosys #1254: [regression] SB_LUT4 mistakenly optimized away

DurandA, 2019-08-06:
I made a simple ring oscillator for iCE40 consisting of SB_LUT4 inverters. After 7129a03083b21c1ed1610126ae28828c56f42369, the SB_LUT4 are optimized away.

To reproduce the issue, the source code with a makefile are available [here](https://gist.github.com/DurandA/8d51cc4e8cf046677284afc9d99352bf). This is a simplified version of @cliffordwolf's [_ringosc.v_](http://svn.clifford.at/handicraft/2015/ringosc/ringosc.v) (for some reason, [_floorplan.v_ oscillator example](https://github.com/YosysHQ/nextpnr/blob/master/ice40/examples/floorplan/floorplan.v) from NextPNR does not oscillate).

Here is a partial diff of the Yosys output:

```diff
@@ -1,5 +1,5 @@
 {
-  "creator": "Yosys 0.8+592 (git sha1 44fd459, clang 3.8.0-2ubuntu4 -fPIC -Os)",
+  "creator": "Yosys 0.8+601 (git sha1 7129a03, clang 3.8.0-2ubuntu4 -fPIC -Os)",
   "modules": {
     "ICESTORM_LC": {
       "attributes": {
@@ -7339,165 +7339,15 @@
             "Q": [ 2 ]
           }
         },
-        "buffers[0]": {
-          "hide_name": 0,
-          "type": "SB_LUT4",
-          "parameters": {
-            "LUT_INIT": 1
-          },
-          "attributes": {
-            "module_not_derived": 1,
-            "src": "top.v:9"
-          },
-          "port_directions": {
-            "I0": "input",
-            "I1": "input",
-            "I2": "input",
-            "I3": "input",
-            "O": "output"
-          },
-          "connections": {
-            "I0": [ 72 ],
-            "I1": [ "0" ],
-            "I2": [ "0" ],
-            "I3": [ "0" ],
-            "O": [ 73 ]
-          }
-        },
-        "buffers[1]": {
-          "hide_name": 0,
-          "type": "SB_LUT4",
-          "parameters": {
-            "LUT_INIT": 1
-          },
-          "attributes": {
-            "module_not_derived": 1,
-            "src": "top.v:9"
-          },
-          "port_directions": {
-            "I0": "input",
-            "I1": "input",
-            "I2": "input",
-            "I3": "input",
-            "O": "output"
-          },
-          "connections": {
-            "I0": [ 73 ],
-            "I1": [ "0" ],
-            "I2": [ "0" ],
-            "I3": [ "0" ],
-            "O": [ 74 ]
-          }
-        },
-        "buffers[2]": {
-          "hide_name": 0,
-          "type": "SB_LUT4",
-          "parameters": {
-            "LUT_INIT": 1
-          },
-          "attributes": {
-            "module_not_derived": 1,
-            "src": "top.v:9"
-          },
-          "port_directions": {
-            "I0": "input",
-            "I1": "input",
-            "I2": "input",
-            "I3": "input",
-            "O": "output"
-          },
-          "connections": {
-            "I0": [ 74 ],
-            "I1": [ "0" ],
-            "I2": [ "0" ],
-            "I3": [ "0" ],
-            "O": [ 75 ]
-          }
-        },
-        "buffers[3]": {
-          "hide_name": 0,
-          "type": "SB_LUT4",
-          "parameters": {
-            "LUT_INIT": 1
-          },
-          "attributes": {
-            "module_not_derived": 1,
-            "src": "top.v:9"
-          },
-          "port_directions": {
-            "I0": "input",
-            "I1": "input",
-            "I2": "input",
-            "I3": "input",
-            "O": "output"
-          },
-          "connections": {
-            "I0": [ 75 ],
-            "I1": [ "0" ],
-            "I2": [ "0" ],
-            "I3": [ "0" ],
-            "O": [ 76 ]
-          }
-        },
-        "buffers[4]": {
-          "hide_name": 0,
-          "type": "SB_LUT4",
-          "parameters": {
-            "LUT_INIT": 1
-          },
-          "attributes": {
-            "module_not_derived": 1,
-            "src": "top.v:9"
-          },
-          "port_directions": {
-            "I0": "input",
-            "I1": "input",
-            "I2": "input",
-            "I3": "input",
-            "O": "output"
-          },
-          "connections": {
-            "I0": [ 76 ],
-            "I1": [ "0" ],
-            "I2": [ "0" ],
-            "I3": [ "0" ],
-            "O": [ 77 ]
-          }
-        },
-        "buffers[5]": {
-          "hide_name": 0,
-          "type": "SB_LUT4",
-          "parameters": {
-            "LUT_INIT": 1
-          },
-          "attributes": {
-            "module_not_derived": 1,
-            "src": "top.v:9"
-          },
-          "port_directions": {
-            "I0": "input",
-            "I1": "input",
-            "I2": "input",
-            "I3": "input",
-            "O": "output"
-          },
-          "connections": {
-            "I0": [ 77 ],
-            "I1": [ "0" ],
-            "I2": [ "0" ],
-            "I3": [ "0" ],
-            "O": [ 78 ]
-          }
-        },
         "buffers[6]": {
           "hide_name": 0,
           "type": "SB_LUT4",
           "parameters": {
-            "LUT_INIT": 1
+            "LUT_INIT": 85
           },
           "attributes": {
             "module_not_derived": 1,
```

daveshah1, 2019-08-07:
The problem is that with `-relut` now enabled by default, it makes the valid and sensible optimisation to convert a chain of 7 NOT gates into a single NOT gate.

It seems like either the relut optimisations should not touch user-instantiated LUTs at all; or they should skip LUTs with some attribute (e.g. `keep`, `dont_touch`, etc) set.

eddiehung, 2019-08-07:
I agree. `-relut` should not touch anything with a `(* keep *)` attached.

DurandA, 2019-08-07:
> The problem is that with `-relut` now enabled by default, it makes the valid and sensible optimisation to convert a chain of 7 NOT gates into a single NOT gate.

Thanks for the explanation!

> It seems like either the relut optimisations should not touch user-instantiated LUTs at all; or they should skip LUTs with some attribute (e.g. `keep`, `dont_touch`, etc) set.

Should I kept this issue open until this behavior is modified?

eddiehung, 2019-08-07:
Thanks for the testcase @DurandA. Please let me know if #1264 helps your situation.

DurandA, 2019-08-08:
> Thanks for the testcase @DurandA. Please let Yme know if #1264 helps your situation.

#1264 works fine with `(* keep="true" *)` attribute. Thank you for addressing this issue.

Would it make sense to add a `--no-relut` switch to yosys?

closed 2019-08-08 by eddiehung

eddiehung, 2019-08-08:
> However, it does not support all caps `(* KEEP *)`.

It looks like Verilog attributes are case-sensitive, so `(* KEEP *)` is different to `(* keep *)`, and it appears that examples from the vendors also recommend lower-case (though I haven't tried if they will accept upper-case).

> Would it make sense to add a --no-relut switch to yosys?

I would say it doesn't make sense, as it `-relut` should always be a neutral or net-benefit optimisation for "typical" designs. It should only fail on atypical structures like combinatorial loops, in which case I would say that attaching `(* keep *)` would be a reasonable workaround.

DurandA, 2019-08-08:
@daveshah1 This is not directly related, but do you know why the same oscillator does not work on the ECP5?

```verilog
module top(input clk, input btn, output [7:0] led);
	// ring oscillator

	wire [6:0] buffers_in, buffers_out;
	assign buffers_in = {buffers_out[5:0], buffers_out[6]};

	(* keep *)
	LUT4 #(
		.INIT(16'd1)
	) buffers [6:0] (
		.Z(buffers_out),
		.A(buffers_in),
		.B(1'b0),
		.C(1'b0),
		.D(1'b0)
	);

	// frequency counter

	reg [23:0] counter = 0;
	always @(posedge buffers_out[6]) begin
		counter <= counter + 1;
	end

	assign led[7] = counter[23];
	assign led[6] = counter[22];
	assign led[5] = counter[21];
	assign led[4] = counter[20];

endmodule
```

daveshah1, 2019-08-08:
ECP5 LUTs are much faster, it is possible that 7 LUTs is too few to get reliable oscillation capable of driving an oscillator.

DurandA, 2019-08-08:
@daveshah1 Even longer chains of <=1001 LUT do not oscillate. I also tried with inverter+delay slices without success:

```verilog
module top(input clk, input btn, output [7:0] led);
	// ring oscillator

	wire [100:0] buffers_in, buffers_out, interlut;
	assign buffers_in = {buffers_out[99:0], buffers_out[100]};

	(* keep *)
	TRELLIS_SLICE #(
		.LUT0_INITVAL(16'd1),
		.LUT1_INITVAL(16'd2)
	) buffers [100:0] (
		.A0(buffers_in),
		.B0(1'b0),
		.C0(1'b0),
		.D0(1'b0),
		.F0(interlut),
		
		.A1(interlut),
		.B1(1'b0),
		.C1(1'b0),
		.D1(1'b0),
		.F1(buffers_out)
	);

	// frequency counter

	reg [23:0] counter = 0;
	always @(posedge buffers_out[100]) begin
		counter <= counter + 1;
	end

	assign led[7] = counter[23];
	assign led[6] = counter[22];
	assign led[5] = counter[21];
	assign led[4] = counter[20];

endmodule
```

PR #1264 (2019-08-08): opt_lut to ignore LUT cells, or those that drive bits, with (* keep *)
Fixes #1254 

Before, and/or without any `(* keep *)`:
```
   Number of cells:                 71
     SB_CARRY                       22
     SB_DFF                         23
     SB_DFFE                         1
     SB_LUT4                        25
```

After:
```
   Number of cells:                 77
     SB_CARRY                       22
     SB_DFF                         23
     SB_DFFE                         1
     SB_LUT4                        31
```
when either of the two `(* keep *)` attributes are present:
 [... cut]